Koenkk/zigbee2mqtt #3342: Hue Motion Sensor (9290012607) does not send illuminance_lux or illuminance

coxtor, 2020-04-12:
# Bug Report
## What happened
My hue motion sensor  (9290012607) does not send illuminance information, all other features work fine.
## What did you expect to happen
The payload contains illuminance information
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:

```
zigbee2mqtt:info  2020-04-12 09:48:09: Update available for 'HueMotion1'
zigbee2mqtt:info  2020-04-12 09:48:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Update available for 'HueMotion1'","meta":{"status":"available","device":"HueMotion1"}}'
zigbee2mqtt:info  2020-04-12 09:52:29: MQTT publish: topic 'zigbee2mqtt/HueMotion1', payload '{"battery":100,"linkquality":68,"occupancy":true,"temperature":20.67,"illuminance":0,"illuminance_lux":0,"update_
available":true}'
zigbee2mqtt:info  2020-04-12 09:52:31: MQTT publish: topic 'zigbee2mqtt/HueMotion1', payload '{"battery":100,"linkquality":70,"occupancy":true,"temperature":20.38,"illuminance":0,"illuminance_lux":0,"update_
available":true}'
zigbee2mqtt:info  2020-04-12 09:52:48: MQTT publish: topic 'zigbee2mqtt/HueMotion1', payload '{"battery":100,"linkquality":68,"occupancy":false,"temperature":20.38,"illuminance":0,"illuminance_lux":0,"update
_available":true}'
zigbee2mqtt:info  2020-04-12 09:58:37: MQTT publish: topic 'zigbee2mqtt/HueMotion1', payload '{"battery":100,"linkquality":65,"occupancy":false,"temperature":20.38,"illuminance":0,"illuminance_lux":0,"update
_available":true}'

```

Koenkk, 2020-04-12:
Can you try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

Seeing that there is an update available, you can also try OTA updating it: https://www.zigbee2mqtt.io/information/ota_updates.html

coxtor, 2020-04-12:
Thanks for your reply, I have tried updating however, the update does not work and times out with: 
` MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Update of 'HueMotion1' failed (Timeout: device did not request any image blocks)","meta":{"status":"update_failed","device":"HueMotion1"}}'
`

is it necessary to enable permit_join to reconfigure? Furthermore to reconfigure do I need to put the device itself into setup mode? Currently I get

`zigbee2mqtt:error 2020-04-12 11:00:36: Failed to configure 'HueMotion1', attempt 1 (Error: Bind 0x0017880104b6bb8c/2 genPowerCfg from '0x00124b0018e1b057/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
`

when reconfiguring.

Other devices on the network are working fine.

Koenkk, 2020-04-12:
Can you try removing the device and repairing it? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

coxtor, 2020-04-13:
I have done so, this is the output:
### removing the sensor:
`zigbee2mqtt:info  2020-04-13 15:27:54: Removing 'HueMotion1'
zigbee2mqtt:info  2020-04-13 15:27:55: Clearing Home Assistant discovery topic for '0x0017880104b6bb8c'
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'homeassistant/binary_sensor/0x0017880104b6bb8c/occupancy/config', payload 'null'
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/temperature/config', payload 'null'
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/illuminance/config', payload 'null'
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/illuminance_lux/config', payload 'null'
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/battery/config', payload 'null'
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/linkquality/config', payload 'null'
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'homeassistant/binary_sensor/0x0017880104b6bb8c/update_available/config', payload 'null'
zigbee2mqtt:info  2020-04-13 15:27:55: Successfully removed HueMotion1
zigbee2mqtt:info  2020-04-13 15:27:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"HueMotion1"}'`

### repairing the sensor
`zigbee2mqtt:info  2020-04-13 15:31:50: Device '0x0017880104b6bb8c' joined
zigbee2mqtt:info  2020-04-13 15:31:50: Starting interview of '0x0017880104b6bb8c'
zigbee2mqtt:info  2020-04-13 15:31:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880104b6bb8c"}}'
zigbee2mqtt:info  2020-04-13 15:31:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880104b6bb8c"}}'
zigbee2mqtt:info  2020-04-13 15:31:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880104b6bb8c"}}'
zigbee2mqtt:info  2020-04-13 15:31:52: MQTT publish: topic 'homeassistant/binary_sensor/0x0017880104b6bb8c/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x0017880104b6bb8c","json_attributes_topic":"zigbee2mqtt/0x0017880104b6bb8c","name":"0x0017880104b6bb8c_occupancy","unique_id":"0x0017880104b6bb8c_occupancy_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0017880104b6bb8c"],"name":"0x0017880104b6bb8c","sw_version":"Zigbee2mqtt 1.12.2","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-04-13 15:31:52: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x0017880104b6bb8c","json_attributes_topic":"zigbee2mqtt/0x0017880104b6bb8c","name":"0x0017880104b6bb8c_temperature","unique_id":"0x0017880104b6bb8c_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0017880104b6bb8c"],"name":"0x0017880104b6bb8c","sw_version":"Zigbee2mqtt 1.12.2","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-04-13 15:31:52: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/illuminance/config', payload '{"unit_of_measurement":"-","device_class":"illuminance","value_template":"{{ value_json.illuminance }}","state_topic":"zigbee2mqtt/0x0017880104b6bb8c","json_attributes_topic":"zigbee2mqtt/0x0017880104b6bb8c","name":"0x0017880104b6bb8c_illuminance","unique_id":"0x0017880104b6bb8c_illuminance_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0017880104b6bb8c"],"name":"0x0017880104b6bb8c","sw_version":"Zigbee2mqtt 1.12.2","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-04-13 15:31:52: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/illuminance_lux/config', payload '{"unit_of_measurement":"lx","device_class":"illuminance","value_template":"{{ value_json.illuminance_lux }}","state_topic":"zigbee2mqtt/0x0017880104b6bb8c","json_attributes_topic":"zigbee2mqtt/0x0017880104b6bb8c","name":"0x0017880104b6bb8c_illuminance_lux","unique_id":"0x0017880104b6bb8c_illuminance_lux_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0017880104b6bb8c"],"name":"0x0017880104b6bb8c","sw_version":"Zigbee2mqtt 1.12.2","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-04-13 15:31:52: MQTT publish: topic 'homeassistant/sensor/0x0017880104b6bb8c/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x0017880104b6bb8c","json_attributes_topic":"zigbee2mqtt/0x0017880104b6bb8c","name":"0x0017880104b6bb8c_battery","unique_id":"0x0017880104b6bb8c_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0017880104b6bb8c"],"name":"0x0017880104b6bb8c","sw_version":"Zigbee2mqtt 1.12.2","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/ [... cut ...]
zigbee2mqtt:info  2020-04-13 15:33:35: MQTT publish: topic 'zigbee2mqtt/0x0017880104b6bb8c', payload '{"battery":100,"linkquality":34,"update_available":false,"occupancy":false,"temperature":23.35,"illuminance":0,"illuminance_lux":0}'
zigbee2mqtt:info  2020-04-13 15:33:44: MQTT publish: topic 'zigbee2mqtt/0x0017880104b6bb8c', payload '{"battery":100,"linkquality":31,"update_available":false,"occupancy":true,"temperature":23.35,"illuminance":0,"illuminance_lux":0}'
`

As you can see the sensor is still not reporting luminance etc. the sensor is new out of the box and worked fine with the standard bridge. What else could be done ?

coxtor, 2020-04-13:
Okay, the repairing made it possible to execute the update. The device is now working as expected. Thank you for the support.

closed 2020-04-13 by coxtor