Koenkk/zigbee2mqtt #3345: UnhandledPromiseRejectionWarning: Error: AREQ - AF

raptr55, 2020-04-12:
Hi 

some times my devices go offline. In the log i see that the device failed to ping and another error:

> **(node:234) UnhandledPromiseRejectionWarning: Error: AREQ - AF** - dataConfirm after 5000ms
>     at Timeout._onTimeout (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
>     at listOnTimeout (internal/timers.js:531:17)
>     at processTimers (internal/timers.js:475:7)
> **(node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 34)**
> (node:234) UnhandledPromiseRejectionWarning: Error: Timeout - 9033 - 11 - 158 - 0 - 1 after 6000ms
>     at Timeout._onTimeout (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
>     at listOnTimeout (internal/timers.js:531:17)
>     at processTimers (internal/timers.js:475:7)
> **(node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 35)**
> zigbee2mqtt:debug 2020-04-12 18:51:31: Failed to ping '0x00124b001d3ac596'
> (node:234) UnhandledPromiseRejectionWarning: Error: Timeout - 40004 - 1 - 159 - 0 - 1 after 6000ms
>     at Timeout._onTimeout (/zigbee2mqtt-1.12.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
>     at listOnTimeout (internal/timers.js:531:17)
>     at processTimers (internal/timers.js:475:7)
> **(node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 36)**
> zigbee2mqtt:info  2020-04-12 18:51:37: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe51f715/availability', payload 'offline'
> **zigbee2mqtt:error 2020-04-12 18:51:37: Failed to ping '0x000b57fffe51f715**'

i'm using the plugin version of zigbee2mqtt for home assistant and the CC2531 Bare Board flashed with the "CC2531_DEFAULT_20190608.zip" firmware and the network has 6 devices

Koenkk, 2020-04-13:
Unfortunately this bug report is missing the required issue template. Please update the OP with the required template (https://github.com/Koenkk/zigbee2mqtt/issues/new).

raptr55, 2020-04-13:
> Unfortunately this bug report is missing the required issue template. Please update the OP with the required template (https://github.com/Koenkk/zigbee2mqtt/issues/new).

my bad, OP updated

Koenkk, 2020-04-13:
Can you also provide the complete herdsman debug loggin?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging (please post a link to pastebin.com, don't put there directly here).

raptr55, 2020-04-13:
> Can you also provide the complete herdsman debug loggin?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging (please post a link to pastebin.com, don't put there directly here).

the logs in the \share\zigbee2mqtt\log\ are different than those present in the add-on page. however i catch the log part [https://pastebin.com/G0dGjVMZ](url) the sensor this time is 0xd0cf5efffe2bd139

raptr55, 2020-04-13:
> > Can you also provide the complete herdsman debug loggin?
> > To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging (please post a link to pastebin.com, don't put there directly here).
> 
> the logs in the \share\zigbee2mqtt\log\ are different than those present in the add-on page. however i catch the log part https://pastebin.com/G0dGjVMZ the sensor this time is 0xd0cf5efffe2bd139

other log portion [https://pastebin.com/QatS8jNR](https://pastebin.com/QatS8jNR)

Koenkk, 2020-04-14:
Can you try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

raptr55, 2020-04-17:
> Can you try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

just flashed but when it starts an error is logged at line 23

[https://pastebin.com/wYgWf8XS](https://pastebin.com/wYgWf8XS)

i will see if the "UnhandledPromiseRejectionWarning: Error: AREQ - AF" shows up again

Koenkk, 2020-04-17:
@raptr55 I checked but that error is not a problem as you use a path to specify the adapter path.

raptr55, 2020-04-17:
> @raptr55 I checked but that error is not a problem as you use a path to specify the adapter path.

got that error again -->[https://pastebin.com/jys0g7A7](https://pastebin.com/jys0g7A7) at line 710

klausdm, 2020-04-19:
I am seen similar error. The zigbee2mqtt complete dies. I cannot restart the service, it cannot fint the stick after this. I ahve to re-connect the stick. What is happening?

https://pastebin.com/WVtaTSCx

klausdm, 2020-04-19:
It's actually enough just to re-connect the stick without restarting zigbee2mqtt service when this happens.

klausdm, 2020-04-21:
Okay, so i re-flashed the coordinator firmware, it solved my issues....

raptr55, 2020-04-28:
@Koenkk news?

Koenkk, 2020-04-29:
@raptr55 I don't know why the `UnhandledPromiseRejectionWarning` happens as the call is awaited correctly. Maybe it has to do with the node version. Can you go inside the zigbee2mqtt container and execute `node --version`?

raptr55, 2020-04-29:
> @raptr55 I don't know why the `UnhandledPromiseRejectionWarning` happens as the call is awaited correctly. Maybe it has to do with the node version. Can you go inside the zigbee2mqtt container and execute `node --version`?

bash-5.0# node --version
v12.15.0

Koenkk, 2020-05-02:
Found the issue, turned out it was a nasty zigbee-herdsman bug. 

Should be fixed in latest dev branch.

raptr55, 2020-07-02:
Hi

I cant install the dev branch at the moment... Hope in the next few months

closed 2020-08-08 by stale[bot]