Koenkk/zigbee2mqtt #3346: LOG updates every 4th second

agoralive, 2020-04-12:
Is there a way to do this less often, or is that a bad idea?

I have 19 paired devices and my log is looks like this every 4th second:

`{"type":"devices","message":[{"ieeeAddr":"0x00124b000be5af5c","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1586724238394},{"ieeeAddr":"0x14b457fffe2e0d2b","type":"EndDevice","networkAddress":51857,"model":"TT001ZAV20","friendly_name":"Temp_tuya","manufacturerID":0,"manufacturerName":"TUYATEC-yg5dcbfu","powerSource":"Battery","modelID":"RH3052","hardwareVersion":1,"dateCode":"20180608","lastSeen":1586723860774},{"ieeeAddr":"0x00158d0002c40efb","type":"EndDevice","networkAddress":47561,"model":"WXKG01LM","friendly_name":"rundknap","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20150424","lastSeen":1586723922834},{"ieeeAddr":"0x00158d000201b9a9","type":"EndDevice","networkAddress":59240,"model":"MCCGQ11LM","friendly_name":"door_1","manufacturerID":4151,"manufacturerName":"LUMI","powerSou...`

I mean it seems like a lot on information very often?

Koenkk, 2020-04-13:
Can you provide more context of the log. This information is typically requested via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigdevicesget

agoralive, 2020-04-13:
That is the entire log, but ofcause it runs every 4 seconds:

Here is the "latest"

`Apr 13 16:00:03 hjemmeflow-server npm[530]: zigbee2mqtt:info  2020-04-13 16:00:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b000be5af5c","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1586786403952},{"ieeeAddr":"0x14b457fffe2e0d2b","type":"EndDevice","networkAddress":51857,"model":"TT001ZAV20","friendly_name":"Temp_tuya","manufacturerID":0,"manufacturerName":"TUYATEC-yg5dcbfu","powerSource":"Battery","modelID":"RH3052","hardwareVersion":1,"dateCode":"20180608","lastSeen":1586786267433},{"ieeeAddr":"0x00158d0002c40efb","type":"EndDevice","networkAddress":47561,"model":"WXKG01LM","friendly_name":"rundknap","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20150424","lastSeen":null},{"ieeeAddr":"0x00158d000201b9a9","type":"EndDevice","networkAddress":59240,"model":"MCCGQ11LM","friendly_name":"door_1","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0x00158d00020075a1","type":"EndDevice","networkAddress":28798,"model":"WXKG01LM","friendly_name":"rundknap_2","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20150424","lastSeen":null},{"ieeeAddr":"0x00158d0001e5aad4","type":"EndDevice","networkAddress":8847,"model":"WXKG01LM","friendly_name":"karina_knap","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20150424","lastSeen":null},{"ieeeAddr":"0x00158d0001e8435a","type":"EndDevice","networkAddress":18903,"model":"WXKG11LM","friendly_name":"firkantet_knap","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch.aq2","hardwareVersion":2,"softwareBuildID":"3000-0001","dateCode":"20161128","lastSeen":null},{"ieeeAddr":"0xccccccfffe2fd70f","type":"EndDevice","networkAddress":11943,"model":"E1525","friendly_name":"ikea_pir1","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"2.0.022","dateCode":"20190308","lastSeen":1586786391905},{"ieeeAddr":"0x000d6ffffec5ea92","type":"EndDevice","networkAddress":26274,"model":"E1743","friendly_name":"ikea_knap_1","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI on/off switch","hardwareVersion":1,"softwareBuildID":"2.0.020","dateCode":"20180921","lastSeen":1586785945334},{"ieeeAddr":"0x000d6ffffedaa9b7","type":"Router","networkAddress":4889,"model":"E1603/E1702","friendly_name":"ikea_220vstik_1","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI control outlet","hardwareVersion":1,"softwareBuildID":"2.0.019","dateCode":"20180828","lastSeen":1586783854408},{"ieeeAddr":"0x14b457fffefc2d61","type":"Router","networkAddress":43883,"model":"E1746","friendly_name":"Ikea_USB_Router","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI Signal Repeater","hardwareVersion":1,"softwareBuildID":"2.2.005","dateCode":"20190311","lastSeen":null},{"ieeeAddr":"0x00158d00033ecdc5","type":"EndDevice","networkAddress":36898,"model":"WSDCGQ01LM","friendly_name":"Temp_ude","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sens","hardwareVersion":30,"softwareBuildID":"3000-0001","dateCode":"20160516","lastSeen":1586786248472},{"ieeeAddr":"0xccccccfffe58fe31","type":"Router","networkAddress":8384,"model":"LED1836G9","friendly_name":"bulb","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 WW 806lm","hardwareVersion":2,"softwareBuildID":"2.1.022","dateCode":"20181203","lastSeen":null},{"ieeeAddr":"0xccccccfffe05c23d","type":"EndDevice","networkAddress":12784,"model":"E1743","friendly_name":"button","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI on/off switch","hardwareVersion":1,"softwareBuildID":"2.2.008","dateCode":"20190410","lastSeen":null},{"ieeeAddr":"0x14b457fffe2e0f3a","type":"EndDevice","networkAddress":61192,"model":"TT001ZAV20","friendly_name":"Temp_tuya2","manufacturerID":0,"manufacturerName":"TUYATEC-yg5dcbfu","powerSource":"Battery","modelID":"RH3052","hardwareVersion":1,"dateCode":"20180608","lastSeen":1586786038604},{"ieeeAddr":"0x00158d0003132ec5","type":"EndDevice","networkAddress":14413,"model":"MCCGQ11LM","friendly_name":"for_dor_sensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","lastSeen":null},{"ieeeAddr":"0x00158d000200694d","type":"EndDevice","networkAddress":65371,"model":"RTCGQ11LM","friendly_name":"xiaomi_pir_1_lamp_kitchen","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_motion.aq2","lastSeen":1586786009884},{"ieeeAddr":"0x00158d0001f9d1e9","type":"EndDevice","networkAddress":29411,"model":"RTCGQ11LM","friendly_name":"pir_lux_1","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_motion.aq2","lastSeen":null},{"ieeeAddr":"0x00158d000227ea9f","type":"EndDevice","networkAddress":59096,"model":"WSDCGQ01LM","friendly_name":"ude_sensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_ht","lastSeen":1586786390823}]}'`

agoralive, 2020-04-13:
So you mean that "something" is requesting this all the time?

Koenkk, 2020-04-13:
@agoralive probably, but for that I need more and debug logging. Please post a more complete log to pastebin.com and link it here.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

agoralive, 2020-04-13:
https://pastebin.com/VVvRrdmB

Koenkk, 2020-04-13:
As can be seen in the log, zigbee2mqtt receives a request to send the device list:

```
Apr 13 20:37:22 hjemmeflow-server npm[513]: zigbee2mqtt:debug 2020-04-13 20:37:22: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '1586803042652'
```

You have to find out who sends this MQTT message

agoralive, 2020-04-13:
Can this be the zigbee2mqtt Node-Red Admin Panel?

Because this is what I am using.....

agoralive, 2020-04-13:
Yes it was.....

Thank you for ALL your help, this was perfect, THANK YOU!!!!!

agoralive, 2020-04-13:
As another question, but you know if this is causing any problems for the MQTT server? Is this too much?

Koenkk, 2020-04-14:
This won't cause any issue for the MQTT server. I will close this issue as the problem has been identified and is not withing zigbee2mqtt.

closed 2020-04-14 by Koenkk