Koenkk/zigbee2mqtt #3347: OTA Update does not stop sending commands

hartmood, 2020-04-12:
I have used used the OTA Update command a couple of days ago. Now I still can seel traffic from this command in the log file. The device is not supported anyway (Xiaomi). 

`Client mosqsub|10992-openhab received PUBLISH (d0, q0, r1, m0, 'zigbee2mqtt/bridge/ota_update/check', ... (8 bytes))`

How can I stop this?

Koenkk, 2020-04-13:
Unfortunately this bug report is missing the required issue template. Please update the OP with the required template (https://github.com/Koenkk/zigbee2mqtt/issues/new).

hartmood, 2020-04-13:
# Bug Report
## What happened
I have used used the OTA Update command a couple of days ago. Now I still can seel traffic from this command in the log file. The device is not supported anyway (Xiaomi).

## What did you expect to happen
The command should have been executed once and then returned a result. Although the server has already been restarted several times, the command is still executed. 

## How to reproduce it (minimal and precise)
It is more about how to stop it than how to reproduce.

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC1352P-2
Adapter firmware version: 20191106

Koenkk, 2020-04-13:
Please attach the debug log as described in the template.

hartmood, 2020-04-13:
[2020_04_13_21.43_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4471529/2020_04_13_21.43_log.txt)

Koenkk, 2020-04-14:
Zigbee2mqtt receives a request to check for an OTA:

```js
zigbee2mqtt:debug 2020-04-13 21:42:23: Received MQTT message on 'zigbee2mqtt/bridge/ota_update/check' with data 'EG_FL_TS'
```

You have to figure out who sends this message.

hartmood, 2020-04-14:
It looks like it just happens when I use the following command in the Terminal:
`mosquitto_sub -d -t zigbee2mqtt/#`
I still have no idea where it comes from as there is most like no service running that executes it.

Koenkk, 2020-04-14:
Probably this is a retained message, read e.g. https://community.openhab.org/t/clearing-mqtt-retained-messages/58221 on how to clear it.

hartmood, 2020-04-14:
Wonderful. That seems to have been it.

closed 2020-04-14 by hartmood