Koenkk/zigbee2mqtt #3352: When setting state+color or brightness+color only state or brightness is applied

sjorge, 2020-04-13:
# Bug Report
## What happened
When calling the set topic with multiple attributes color gets applied first, this does not work for most bulbs (except trådfri)

## What did you expect to happen
I except the bulb to be turned on first, before setting the color.

## How to reproduce it (minimal and precise)
The bulb is has state=off (was warm white) when sending this, both result in the same problem but I tried both to be sure.
```
# with brigtness
mosquitto_pub -h localhost -p 1883 -t zigbee2mqtt/bedroom/desk_lamp/set -m '{"brightness_percent": 50, "color":"#FF0000"}'
# or just with state
mosquitto_pub -h localhost -p 1883 -t zigbee2mqtt/bedroom/desk_lamp/set -m '{"state": "on", "color":{"x":0.4573,"y":0.41}}'
```

Result (and annotation)
```
zigbee2mqtt:debug 2020-04-13 19:12:57: Received MQTT message on 'zigbee2mqtt/bedroom/desk_lamp/set' with data '{"brightness_percent": 50, "color":"#FF0000"}'
```
Nice, order (brightness/state before color in payload) is remembered but it doesn't seem to matter.

```
zigbee2mqtt:debug 2020-04-13 19:12:57: Publishing 'set' 'color' to 'bedroom/desk_lamp'
zigbee2mqtt:info  2020-04-13 19:12:57: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp/bulb', payload '{"state":"OFF","linkquality":0,"brightness":255,"color_temp":369,"color":{"x":0.7006,"y":0.2993,"saturation":79,"hue":34},"color_mode":1}'
zigbee2mqtt:info  2020-04-13 19:12:57: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"OFF","brightness":255,"color":{"x":0.7006,"y":0.2993,"saturation":79,"hue":34},"color_temp":369}'
```
Color gets pushed first, but the bulb is off... so it is ignored... both the device and group get opportunistically update in the publish mqtt payload.

```
zigbee2mqtt:debug 2020-04-13 19:12:57: Publishing 'set' 'brightness_percent' to 'bedroom/desk_lamp'
zigbee2mqtt:info  2020-04-13 19:12:57: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp/bulb', payload '{"state":"ON","linkquality":0,"brightness":127,"color_temp":369,"color":{"x":0.7006,"y":0.2993,"saturation":79,"hue":34},"color_mode":1}'
zigbee2mqtt:info  2020-04-13 19:12:57: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","brightness":127,"color":{"x":0.7006,"y":0.2993,"saturation":79,"hue":34},"color_temp":369}'
zigbee2mqtt:debug 2020-04-13 19:12:58: Received Zigbee message from 'bedroom/desk_lamp/bulb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
```
Bulb reports that it switched to on.

```
zigbee2mqtt:debug 2020-04-13 19:12:58: Received Zigbee message from 'bedroom/desk_lamp/bulb', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":127}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-13 19:12:59: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","brightness":127,"color":{"x":0.7006,"y":0.2993,"saturation":79,"hue":34},"color_temp":369}'
zigbee2mqtt:info  2020-04-13 19:12:59: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp/bulb', payload '{"state":"ON","linkquality":0,"brightness":127,"color_temp":369,"color":{"x":0.7006,"y":0.2993,"saturation":79,"hue":34},"color_mode":1}'
```

Bulb also reports the brightness was set to 50% (as expected)

Sending state or brightness to /set, then sending color to /set works... but sadly I cannot control the grouping of the sending source :(

## Debug Info
Zigbee2mqtt version: `1.12.2-dev (commit #bd5cb77)`
Adapter hardware: CC2531
Adapter firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":
0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20200410}}`

sjorge, 2020-04-13:
I had a poke around but it is not immediately clear what (z2m, herdsman, or converters) that is deciding the ordering.

Koenkk, 2020-04-14:
I expect adding `brightness_percent` to this list fixes it: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/entityPublish.js#L135

sjorge, 2020-04-14:
I will try tonight and report back... prob with a PR if it works.

sjorge, 2020-04-14:
I wonder if we should make this more intellegent, state/brightness/brightness_percent should be FIRST... unless state==OFF brightness*==0

sjorge, 2020-04-14:
Original issue fixed due to merge

closed 2020-04-14 by sjorge