Koenkk/zigbee2mqtt #3353: Gledopto GL-C008 no linkquality

myradon, 2020-04-13:
# Bug Report

## What happened
None of my Gledopto GL-C-008's show any linkquality anymore. 

## What did you expect to happen
Inspecting topic. I would assume I even see linkquality-key in JSON-payload. It's missing. 
`{"state":"ON","color":{"x":0.54,"y":0.393},"color_temp":356,"brightness":89}`

Complete logfile https://privatebin.net/?9bbacc1dc518aed9#qRSathSEDWmpqXNhzRCA3H2YgSnv3oBLSx4twhAx4kr

_you can search on friendly devices;

- light_cove_bedroom
- light_cove_hall
- light_shelf_kitchen_

## How to reproduce it (minimal and precise)
I can't reproduce it cause its absence is the issue.  

## Debug Info
Zigbee2mqtt version:  1.12.0 (commit #840b9d9)
Adapter hardware: CC2531 
Adapter firmware version:  '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

Koenkk, 2020-04-14:
Bulbs normally don't send any messages to the coordinator, therefore we don't know the linkquality. You can achieve it by e.g. executing a get by publishing to `zigbee2mqtt/FRIENDLY_NAME/get` payload `{"state": ""}`

closed 2020-04-14 by Koenkk

myradon, 2020-04-14:
Hi, Why does the Gledopto has a linkquality sensor? Is it obsolete? The docs show that sensor. 

Some of the same Hue Bulbs show linkquality in history while the other doesn't. 1 Hue Iris has linkquality key while the other hasn't. Almost all IKEA leds show linkquality. I don't get the behaviour.

Koenkk, 2020-04-14:
Because all devices potentially can have an LQI