Koenkk/zigbee2mqtt #3354: Zigbee2MQTT crash after adding unsupported device

macdelft, 2020-04-13:
## Bug Report

When pairing my SR-ZG9080A Curtain Motor Controller, Zigbee2MQTT crashes

## What did you expect to happen

The device seems to pair successfully (Added to configuration.yaml) however, Zigbee2MQTT crashes in the process. 
Pair seems to be successful;

`  zigbee2mqtt:info 4/13/2020, 10:19:17 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xccccccfffe269b0e","meta":{"modelID":"Motor Controller"}}'
`

## Logs

Zigbee2MQTT debug-mode logging
[https://pastebin.com/embed_js/Y7uZu403](url)

Zigbee-Herdsman Debug logs:
[https://pastebin.com/embed_js/8AT1wp3y](url)

## How to reproduce it (minimal and precise)

Factory reset SR-ZG9080A, this enables pairing mode to a coodinator network. Zigbee2MQTT picks up the device, tries to pair/pairs and crashes

## Debug Info
Zigbee2mqtt version: 1.6.0
Adapter hardware: CC2531
Adapter firmware version: Unknown

Koenkk, 2020-04-14:
1.6.0 is quite an old version, make sure you are on the latest zigbee2mqtt first (1.12.2)

macdelft, 2020-04-15:
Hi @Koenkk 
It's resolved after upgrading to the latest version. Sorry for disturbing!

closed 2020-04-15 by macdelft