Koenkk/zigbee2mqtt #3355: Fytur Blinds Largely Unreliable

jdtsmith, 2020-04-14:
I've been trying to make zigbee2mqtt work with a pair of Fyrtur blinds and my CC2531 for the better part of half a year, but sadly, I'm actually worse off now than when I started.  I've worked at the bleeding edge of the dev branch, on stable master, Z1.2 stack, now on Z3.0.X stack, you name it.  I've upgraded device firmware.  Really explored pretty much every variable I can think of.  So I'm writing to see if anyone can offer experience on the topic.

The issues I have:

1. The blinds randomly drop off the network after less than one day, leaving a message like:
```
Apr 13 21:18:02 screenpi npm[6524]: zigbee2mqtt:warn  2020-04-13 21:18:02: Device '0xd0<snip>' left the network
Apr 13 21:18:02 screenpi npm[6524]: zigbee2mqtt:info  2020-04-13 21:18:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xd
Apr 13 21:18:08 screenpi npm[6524]: zigbee2mqtt:warn  2020-04-13 21:18:08: Device '0xd0cf<snip>' left the network
Apr 13 21:18:08 screenpi npm[6524]: zigbee2mqtt:info  2020-04-13 21:18:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xd
```
2. I can bind the blinds to an ikea up/down remote, and it (usually) can control them as a group, but it cannot stop or redirect the blinds once moving.
3. I can't command a group of Fytur blinds to do anything.

Has anyone had better luck than me with these blinds?  Is anyone using them successfully with z2m?  I have a few other end devices, like Aqara motion + illuminance sensor, and they remain reliably attached to the network.

swissbyte, 2020-04-14:
Hi 

Some days ago i had also stability issues with my cc2531 stick. See here: #3330 its not absolutley the same problem than yours, but after flashing this firmware: 
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
everything went good. 

But your message: 
`Apr 13 21:18:02 screenpi npm[6524]: zigbee2mqtt:warn  2020-04-13 21:18:02: Device '0xd0<snip>' left the network`
Indicates that the blind itselfs disconnects itself from the network.

jdtsmith, 2020-04-19:
Yes it is strange.  I re-paired them and this time just did not even attempt to group or bind to an IKEA up/down remote, and they have stayed attached to the network for about a week.  It does seem to me that others have had success with grouping and remote-binding groups of these blinds, and the deconz people seem to have them working perfectly.  So feels like we've still got some progress to make...

closed 2020-06-28 by stale[bot]