Koenkk/zigbee2mqtt #3358: Zigbee2mqtt won't start with CC2531 ZBOSS firmware

LeventeKocsis, 2020-04-14:
Hi,

I tried to flash the CC2531 with ZBoss firmware, but zigbee-herdsman won't start. If I flash with CC2531ZNP-Prod firmware it works, but I cannot start de ZBoss packet sniffer.
Can be reproduced with latest Zigbee2mqtt version, on Windows 10.

PS: Is there any other utility for packet sniffing on Windows operating system?

swissbyte, 2020-04-14:
As far as i know, zigbee2mqtt is not developed to be used together with ZBoss packet sniffer FW. If you want to do sniffing you must use for example wireshark together with the ZBoss utility. Take a look here: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

You need a tool "ZBoss_sniffer.exe" on windows for example. 
I have already done this successfully. 

Tell me whether it is working or not.

LeventeKocsis, 2020-04-14:
I have only one CC2531, with Coordinator firmware CC2531ZNP-Prod.hex. In the documentation, which you have sent to me, for Windows I need to flash the CC2531 with ZBoss, otherwise it's not working the ZBoss_sniffer.exe.
After starting ZBoss_sniffer.exe, I'm getting access denied on the serial port (mqtt-herdsman).

I've made a small video, where you can see my tests:
https://drive.google.com/file/d/1J3lKi63sx9U0UT2xt4ZiGLGsUWIS1Ie1/view

Thank you

swissbyte, 2020-04-14:
OK i see your problem. 
You dont need zigbee2mqtt at all for sniffing. 
ZBoss is not related to zigbee2mqtt at all. 
You need to flash the CC2531 Stick with the ZBoss firmware as you already did. 
Then open the ZBoss_sniffer.exe and choose the correct com port. 
Thats it. You dont need zigbee2mqtt. 

ZBoss_sniffer.exe will communicate with the ZBoss sniffer firmware on the cc2531 stick and make them available to wireshark.

LeventeKocsis, 2020-04-14:
Ok, thank you.
In this case do I need a zigbee router (coordinator) which communicates with the end device?

swissbyte, 2020-04-14:
If you want to sniff the communication between your device and zigbee2mqtt then you will need 2 CC2531 devices (sticks). One with zigbee2mqtt firmware and one with ZBoss firmware.

LeventeKocsis, 2020-04-15:
Thank you for the clarifications.

Best regards.

closed 2020-04-15 by LeventeKocsis