Koenkk/zigbee2mqtt #3359: [IDEA!] Pairing stick

swissbyte, 2020-04-14:
Hi there

This is a new idea and not an issue. 
I thought about cloning an existing CC2531 stick to use it as a "pairing device". 
Idea is to power the stick with, lets say, a powerbank and then go to the device which should be fershly paired. now the start the pairing session. The device will pair with the CC2531 stick since due to the lower power during pairing it is not able to reach the real ZNP. After the pairing was successfull, the freshly paired device will be able to connect to the real network since it has all necessary information "NWK" etc. 

Would this be possible? 
Thanks.

Koenkk, 2020-04-14:
- Why not just use a router for this? 
- When the device is out of range, how can it work when the "pairing stick" is powered off again?

swissbyte, 2020-04-14:
Ok, that means if i flash a CC2531 stick with the router firmware, pair the stick with my ZNP, then go with my stick to my device which should get freshly paired it can pair with the router?

I am asking this because i have some zigbee switches inside the wall behind my light switches, and it would be extremly tedious to take my ZNP to all the switches... But a stick with a battery would be no problem. The device itself is not out of range of the regular ZNP. But, as you know, zigbee devices lower their transmit power during the pairing for security reasons. Therefore the device would not be able to reach my ZNP during the first pairing.

Koenkk, 2020-04-15:
Yes that's possible indeed. I'm not aware of the lowered transmission power while pairing.

swissbyte, 2020-04-16:
Great! I will try that. Yes, this lowered transmission power is part of the Zigbee specification. It should help to keep your NWK secret. As you know, at the beginning, all ZLL product communicate during pairing with the same, public NWK. After this first connection to the ZNP, the ZNP shares the "real" NWK with the device which wants to join the network. During this time, one with a sniffer could easily catch the NWK. Sure, the NWK is transmited encrypted, using the hard-coded ZLL NWK, but since a few years, this NWK is public. 

This is why the devices "must" send with the lowest possible power to prevent others to listen to the key.

tsprivate, 2020-04-25:
Just a side-note here. I'm using the cheap [IKEA repeater](https://www.zigbee2mqtt.io/devices/E1746.html) plugged into my battery-powerbank and can than pair new devices everywhere in the house easily

swissbyte, 2020-04-28:
Great, thank you for your reply

closed 2020-04-28 by swissbyte