Koenkk/zigbee2mqtt #3362: Allow runtime customization in the `Dockerfile`

pranasziaukas, 2020-04-14:
I would like to customize some configuration when launching the `zigbee2mqtt` service.

However, at the moment this is not possible as the `ENTRYPOINT`
https://github.com/Koenkk/zigbee2mqtt/blob/911cd58ef27e071d20f48249fd16d53b89ac888d/docker/Dockerfile#L30
bootstraps and starts the server, and does not accept any arguments.

I'd suggest the following change to the `Dockerfile`
```
ENTRYPOINT ["./run.sh”]
CMD [“npm”, “start"]
```
along with `exec "$@"` at the end of `./run.sh`. It would allow `CMD` to be overridden.

Also, I think it adheres to the best practices, as outlined in the Docker documentation [here](https://docs.docker.com/engine/reference/builder/#exec-form-entrypoint-example). Happy to submit a PR.

Koenkk, 2020-04-14:
No problem, please create a PR

Koenkk, 2020-04-15:
Closing this as PR got merged

closed 2020-04-15 by Koenkk