Koenkk/zigbee2mqtt #3365: Add siren trigger HeiMain HS2WD-E to automation

Nickfrede, 2020-04-14:
Hi all,
Noob question - I searched for an answer on GitHub and Online but I just could not get a straight answer.
I am able to trigger the allarm of the HeiMain HS2WD-E and stop it in Home Assistant - Developer Tools - MQTT
**Publish a packet**
 with the following details ito the **topic** field:
_zigbee2mqtt/0x000d***********set_
and those details into the **Payload (template allowed)** field:
{"warning": {"duration": 1, "mode": "emergency", "strobe": true}}
[image]

But still I have not manage to add an action "mqtt.Publish"to my existing allarm automation to trigger the siren.
I used the configuration section of Home Assistant, but I cannot figure out how to correclty format  the message using the UI:
[image]

Ideally I would like to also create a script/button on the UI to turn on/off the siren.
Thanks in advance for your help!

swissbyte, 2020-04-15:
Hi Nickfrede

This question is not directly related to zigbee2mqtt. You could find help in the Home Assistant community. Checkout their discord chat: https://www.home-assistant.io/help/

Thanks

JLFN, 2020-04-16:
This work good for me.
````.yaml
script:
  siren_alarm_on:
    alias: 'zigbee siren on'
    sequence:
      - service: mqtt.publish
        data_template:
          topic: "zigbee2mqtt/0x00000000/set"
          payload: '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
#
  siren_alarm_off:
    alias: 'zigbee siren off'
    sequence:
      - service: mqtt.publish
        data_template:
          topic: "zigbee2mqtt/0x000d00000/set"
          payload: '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
#
switch:
  - platform: "mqtt"
    name: Heiman siren
    state_topic: "zigbee2mqtt/0x000d0000/set"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
    payload_on: '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
    # value_template: "{{ value_json.state }}"
    command_topic: "zigbee2mqtt/0x0000000/set"
```

Nickfrede, 2020-04-20:
> This works good for me.
> 
> ```yaml
> script:
> #
>   siren_alarm_on:
>     alias: 'zigbee siren on'
>     sequence:
>       - service: mqtt.publish
>         data_template:
>           topic: "zigbee2mqtt/0x00000000/set"
>           payload: '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
> #
>   siren_alarm_off:
>     alias: 'zigbee siren off'
>     sequence:
>       - service: mqtt.publish
>         data_template:
>           topic: "zigbee2mqtt/0x000d00000/set"
>           payload: '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
> #
> switch:
>   - platform: "mqtt"
>     name: Heiman siren
>     state_topic: "zigbee2mqtt/0x000d0000/set"
>     availability_topic: "zigbee2mqtt/bridge/state"
>     payload_off: '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
>     payload_on: '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
>     # value_template: "{{ value_json.state }}"
>     command_topic: "zigbee2mqtt/0x0000000/set"
> ```
> ```

It works! Thanks a lot for the help!

closed 2020-04-20 by Nickfrede

xMariusz, 2020-08-19:
@Nickfrede How did you pair the HS2WD-E? Unfortunately, I have a problem with connecting to zigbee2mqtt

otakon87, 2021-02-08:
> How did you pair the HS2WD-E? Unfortunately, I have a problem with connecting to zigbee2mqtt

did you manage to solve it? I'm having the same problem I think, zigbee2mqtt doesn't even see the siren, even though it's in pairing mode (green light flashing)

Nickfrede, 2021-02-08:
All I did what to copy and paste the script @JLFN  wrote in this thread and it worked

otakon87, 2021-02-08:
I'm actually not able to pair it with zigbee2mqtt, tried even the edge version. It doesn't appear at all in the logs, this is the first time it happens to me with a zigbee device.

jamesckc7, 2022-04-08:
> This works good for me.
> 
> ```miniyaml
> script:
> #
>   siren_alarm_on:
>     alias: 'zigbee siren on'
>     sequence:
>       - service: mqtt.publish
>         data_template:
>           topic: "zigbee2mqtt/0x00000000/set"
>           payload: '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
> #
>   siren_alarm_off:
>     alias: 'zigbee siren off'
>     sequence:
>       - service: mqtt.publish
>         data_template:
>           topic: "zigbee2mqtt/0x000d00000/set"
>           payload: '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
> #
> switch:
>   - platform: "mqtt"
>     name: Heiman siren
>     state_topic: "zigbee2mqtt/0x000d0000/set"
>     availability_topic: "zigbee2mqtt/bridge/state"
>     payload_off: '{"warning": {"duration": 0, "mode": "stop", "strobe": false}}'
>     payload_on: '{"warning": {"duration": 1800, "mode": "emergency", "strobe": true}}'
>     # value_template: "{{ value_json.state }}"
>     command_topic: "zigbee2mqtt/0x0000000/set"
> ```
> ```

sorry, I am new to HA.  Just wondering where should I put those information? in zigbee2mqtt/configuration.yaml  or somewhere else? thank you

Sarunas88, 2022-05-17:
> All I did what to copy and paste the script @JLFN wrote in this thread and it worked

hey, can you tell me where you put those line in ? lots of people having problems with this sirene and they dotn get it what to do, me too

Sarunas88, 2022-05-18:
> 

did you find it out ?

Sarunas88, 2022-05-18:
> sorry, I am new to HA. Just wondering where should I put those information? in zigbee2mqtt/configuration.yaml or somewhere else? thank you

have you figured it out ?

snoopfab, 2023-01-19:
> > 
> 
> can you please say where to post this line, sitting for 3 days on this already :(

In your home assistant configuration file (configuration.yaml)