Koenkk/zigbee2mqtt #3367: [docker] z2m not starting correctly

tavalin, 2020-04-15:
# Bug Report
## What happened
Just pulled today's `lastest-dev` Docker image on my RPi4 and z2m no longer starts up.

The only output in my Docker log is the following:

```
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
```

Looking at the recent commits, it may be something to do with this commit https://github.com/Koenkk/zigbee2mqtt/commit/d19367391767db1dd20c06d2217b5836980fbbf2

Pulling the Docker image `latest` means that z2m starts correctly so it is definitely a recent commit that has caused problems.

## What did you expect to happen

z2m to start correctly

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

Koenkk, 2020-04-15:
For me it still works, can you provide your docker compose or run command?

fnxpt, 2020-04-15:
Just updated to the lastest dev and I have the same issue

fnxpt, 2020-04-15:
reverted to master branch and its working fine

tavalin, 2020-04-16:
@Koenkk I use Portainer but I will try and see if I can output the run command.

Koenkk, 2020-04-17:
@fnxpt please provide the necessary information: https://github.com/Koenkk/zigbee2mqtt/issues/3367#issuecomment-614184294

fnxpt, 2020-04-17:
Like tavalin im running it with portainer, but not doing anything special, just defining the device and the volume... Im going to try enable debug mode to check if I can get more details

fnxpt, 2020-04-17:
I was actually running in debug mode, and no log directory is created when start the service, is there any env or parameter I can specify on the entry-pint

fnxpt, 2020-04-17:
Removed all the files from the volume and a new configuration file was created... but I still get the same service crash.
Using '/app/data' as data directory
Creating configuration file...
Using '/app/data' as data directory
Using '/app/data' as data directory

fnxpt, 2020-04-17:
Changed the entrypoint from ./run.sh to a command "npm start" and I see more stuff on logs... now its failing due to another issue Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

fnxpt, 2020-04-17:
Now Im getting this... Error: AREQ - ZDO - stateChangeInd after 60000ms
I already cleanup everything but I get the same error... any suggestions @Koenkk

fnxpt, 2020-04-17:
I manage to bypass the stateChangeInd error... my conclusion is... if we run ./run.sh as entrypoint it doesnt work... if we force the command to "npm start" it works

fnxpt, 2020-04-17:
Maybe something in this commit https://github.com/Koenkk/zigbee2mqtt/commit/d19367391767db1dd20c06d2217b5836980fbbf2

Koenkk, 2020-04-17:
This command shouldn't be needed as in case no command is provided it should just execute `npm start`.

fnxpt, 2020-04-17:
Im not sure, if the recreate the container user portainer, it will re-use the same configurations, for me command was empty this was making the last line of the run.sh exec nothing.

Koenkk, 2020-04-18:
I've reverted the change that caused this issue so closing this.

closed 2020-04-18 by Koenkk