Koenkk/zigbee2mqtt #3369: Add xbee API

alekslyse, 2020-04-16:
Would it be impossible to implement the npm API for xbee in this project

https://www.npmjs.com/package/xbee-api

Koenkk, 2020-04-16:
For what purpose? For usage as a coordinator?

alekslyse, 2020-04-16:
Xbee can do much more than just being a router. It's like an ZigBee version of a nodemcu so you can use gpio on them, run local python code,  and interconnect features like SMS etc. The API will make it easier to use xbee as an advance ZigBee multi-purpose device without having to rewrite the communication protocol. The alternative is to try and write the ZigBee commands directly in the to and from ZigBee files, but if the wheel is invented why not use the existing code

alekslyse, 2020-04-16:
My goal is to be able to use more xbee commands like https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/xbee/xbee_io.py

alekslyse, 2020-04-16:
But if it's a zcl API already integrated in z2m I guess I can use that?

Koenkk, 2020-04-17:
@alekslyse it looks that this code send ZCL on/off commands to certain endpoints (in toZigbee this is done with e.g.: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L122). Once you figure out what commands and endpoints these commands need to be send to I can help you implementing that in a toZigbee converter.

closed 2020-06-26 by stale[bot]