Koenkk/zigbee2mqtt #3370: Z2M won't start after update, old config and database restored

arjenanker, 2020-04-16:
I recently migrated from a CC2531 to a CC26X2R1. Changed configuration.yaml, different pan_id, re-paired all of my devices, had to change a group because a device had a different group id after the migration and that was that, all worked beautifully and much more stable.

So this morning I decided to run an update of zigbee2mqtt, for no particular reason beside staying up to date. Update ran and zigbee2mqtt wasn't running. Started the service, not running. Looked at the logs... device not found, the CC2531 wasn't found. The one that wasn't even supposed to be in my configuration.

Turns out, the update somehow restored my old config and database and everything. Inside the data dir is another data dir, containing the right configuration... however, if I copy that to the data directory, z2m won't start either....

`arjen@domoticz:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.12.2-dev start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-04-16 08:50:49: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-04-16.08-50-49' filename: log.txt
zigbee2mqtt:debug 2020-04-16 08:50:49: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-04-16.08-16-08'
zigbee2mqtt:debug 2020-04-16 08:50:50: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-04-16 08:50:50: Starting zigbee2mqtt version 1.12.2-dev (commit #07dc837)
zigbee2mqtt:info  2020-04-16 08:50:50: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-04-16 08:50:50: Using zigbee-herdsman with settings: '{"network":{"panID":6757,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-Texas_Instruments_XDS110__03.00.00.07__Embed_with_CMSIS-DAP_L1100AFO-if00"},"adapter":{"concurrent":null}}'
zigbee2mqtt:error 2020-04-16 08:51:51: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-16 08:51:51: Failed to start zigbee
zigbee2mqtt:error 2020-04-16 08:51:51: Exiting...
zigbee2mqtt:error 2020-04-16 08:51:51: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.2-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.2-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/arjen/.npm/_logs/2020-04-16T06_51_51_395Z-debug.log`

Going to attempt a reboot of the whole machine, because it seems something keeps the serial port occupied, even through a reboot of the VM...

Koenkk, 2020-04-16:
- Does downgrading to 1.12.2 fix the problem?
- Can you provide the herdsman debug logging when starting with 1.12.2-dev?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

arjenanker, 2020-04-16:
I've managed to get it working again, mostly...

Started first with an empty data dir, only the configuration.yaml. That worked (but obviously all devices were gone). Stopped z2m and added the database.... that ran, but the devices wouldn't work. Had to re-pair most, one started working on it's own. It's weird. I'll get back to you if I manage to get everything working again.

arjenanker, 2020-04-16:
Well, got everything working again. Had to pair everything all over again, except for 1 remote, strangely enough. So groupID's changed, had to reconfigure that too. Real pain and no clue how it could have happened.

I use the update script from the wiki:
`# Stop zigbee2mqtt and go to directory
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Backup configuration
cp -R data data-backup

# Update
git checkout HEAD -- npm-shrinkwrap.json
git pull
npm ci

# Restore configuration
cp -R data-backup/* data
rm -rf data-backup

# Start zigbee2mqtt
sudo systemctl start zigbee2mqtt
`
So I see no way how the configuration could have been overwritten with old information.

Koenkk, 2020-04-16:
I just fixed a critical bug in the latest dev branch (which was only present in zigbee2mqtt 1.12.2-dev) breaking startup with freshly flashed adapters. 

If you don't experience this issue anymore, please close it.

arjenanker, 2020-04-16:
I've got it fixed. I'll update again and close this.

closed 2020-04-16 by arjenanker

coolguymatt, 2020-04-18:
If I am having this issue is the only solution repairing all devices?