Koenkk/zigbee2mqtt #3371: Suddenly can't publish?

ownaginatious, 2020-04-16:
# Bug Report
## What happened

Everything was working fine, and then suddenly I was unable to publish any events to control any of my devices. I still receive messages from all of them though. I only have around 10 devices total. I literally didn't change anything :|

## What did you expect to happen

Everything to continue working as normal.

## How to reproduce it (minimal and precise)

Not sure, it just happened randomly. I tried reflashing the firmware on my coordinator twice, which didn't make any difference. I've attached logs. I don't think it's a power issue as I'm running this off of a desktop computer, which should be able to provide enough current. I live in a pretty modestly sized apartment, so there definitely shouldn't be any range issues.

I was running 1.12.0, but then switched to 1.12.2 in hopes of maybe fixing something. Unfortunately, that didn't help.

I'm running this on FreeBSD 12.1 inside of a jail.

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

I've attached logs [here](https://pastebin.com/14szpafh) when starting `zigbee2mqtt` and attempting to toggle some of my devices. All of them either consistently timeout or crash with no network route.

Any help as to what could be going on here would be greatly appreciated as it's kind of hard not being able to control any of my things anymore :sweat_smile:

ownaginatious, 2020-04-16:
So I tried starting from a fresh config and repaired two switches. They worked for about 5 minutes, and then stopped again with error messages like this every time I try to set the state:

```
igbee2mqtt:error 2020-04-16 07:49:11: Publish 'set' 'state' to '0x000d6ffffee42357' failed: 'Error: Command 0x000d6ffffee42357/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":17018,"options":0,"radius":30}' failed with status '199' (expected '0'))'
zigbee2mqtt:info  2020-04-16 07:49:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x000d6ffffee42357' failed: 'Error: Command 0x000d6ffffee42357/1 genOnOff.off({}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: SREQ '--> ZDO - extRouteDisc - {\"dstAddr\":17018,\"options\":0,\"radius\":30}' failed with status '199' (expected '0'))'","meta":{"friendly_name":"0x000d6ffffee42357"}}'
```

Anyone know if this sounds like defective hardware?

Koenkk, 2020-04-17:
It complain about a network table being full, I would suggest switching to the source routing firmware which has more space for this: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

ownaginatious, 2020-04-17:
@Koenkk thanks! I actually tried switching to that yesterday shortly after posting this, and it seems to have resolved the issue 👍 Curious though, how can you tell from the logs that the network table is full?

Koenkk, 2020-04-17:
`failed with status '199' (expected '0'))'` -> error 199 -> 199 in hex is 0xC7 -> now check https://www.silabs.com/content/usergenerated/asi/cloud/attachments/siliconlabs/en/community/wireless/zigbee-and-thread/forum/jcr:content/content/primary/qna/etrx357-lr_cvs_texas-TmdR/Z-Stack%20ZNP%20Interface%20Specification.pdf page 20 -> ZnwkTableFull 0xc7

ownaginatious, 2020-04-21:
Great, I'll consult this spec doc first next time I have issues. Thanks!

closed 2020-04-21 by ownaginatious