Koenkk/zigbee2mqtt #3372: IKEA Trafdri Dimmers ICTC-G-1 not binding to lamps after OTA update

andrewsweet4, 2020-04-16:
# Bug Report
## What happened
Yesterday i OTA updated 2 of my Tradfri ICTC-G-1 dimmers to the latest firmware. After the update the dimmer stopped controlling the paired IKEA lamp (My setup consists of 4 Tradfri LED1623G12 lamps and 4 Tradfri ICTC-G-1 dimmers). I can see the dimmers sending data through home assistant, and can also control the lamps from there individually. But if i try to pair one of the updated dimmers to a lamp through mqtt, it seems that the bind was successful in the logs, but the lamp doesn't respond. If i try to manually pair the dimmer to the lamp it succeeds, but many of the other lamps also get paired to it despite being in different rooms or even have their power switched off during the pairing proccess! I tried the same thing (binding a dimmer to a lamp through mqtt command) with a dimmer with older firmware, and everything works fine, so i suspect that the new ICTC-G-1 dimmer firmware changed something. 
Remark: i was able to reproduce the problem multiple times by factory resetting all my devices and deleting all the zigbee2mqtt configuration and starting from scratch. (pairing the coordinator to the devices and trying to pair the dimmers to the bulbs both manually and through mqtt)

## What did you expect to happen
I expect the dimmers flashed with the latest firmware to be able to get paired with bulbs through the usual zigbee2mqtt mqtt process.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608
**Dimmers with new firmware:** 
hardwareVersion:30
softwareBuildID: 2.3.028 
dateCode: 20191014 
**Dimmers with old firmware:** 
hardwareVersion:30
softwareBuildID: 1.1.1.1-5.7.2.0
dateCode: 20160923

nldroid, 2020-04-16:
Just a wild guess: did you try unbinding them first? Maybe they're bound to a default group after updating..

andrewsweet4, 2020-04-16:
Yes i unbound them manually and olso tried resetting ALL my devices to factory defaults and paired them again with the coordinator one by one. I also tried adding each of my bulbs to a different group to be sure they are sepperated from each other with the same results. Even if i had forgot a device bound to another one, wouldn't this bond be erased/forgot after factory resetting the devices?

didiht, 2020-04-16:
In my opinion the handling of the dimmer changed with the new firmware in the same way as for the remote https://www.zigbee2mqtt.io/information/binding.html#binding-a-remote-to-a-group.

This means: 

- create a new group
- bind the dimmer to the new group (maybe an unbind is necessary before)
- add bulbs to this group

andrewsweet4, 2020-04-17:
> In my opinion the handling of the dimmer changed with the new firmware in the same way as for the remote https://www.zigbee2mqtt.io/information/binding.html#binding-a-remote-to-a-group.
>
> This [...]

Hello
I did what you said with somewhat of success, but i still experience some strange behaviour. Here is what i did:
Created a new group, binded a dimmer to this group through mqtt commands and added 2 IKEA tradfri bulbs to the group. Everything works fine here. I can see the dimmers and control the lights through home assistant and also control these 2 lights directly from the dimmer.
But here comes the catch! i Created another new group as before, binded another dimmer to the group and added another single light bulb to the group. The exact procedure as before but with only one bulb in the group. In the logs i can see that the dimmer was successfully paired to the group, and i can see both the bulb and dimmer through home assistant, but the bulb does not respond to the newlly paired dimmer! This is frustrating! Note that both dimmers used in this experiment have the same new firmware flashed, and all the bulbs have the same firmware as well.

didiht, 2020-04-17:
Did you check which groupId is indeed sent from the dimmer?

You see this in the logs with level "debug" when moving the dimmer, as I remember.

andrewsweet4, 2020-04-17:
Yes the group ID that the dimmer sends is correct.

[image]

didiht, 2020-04-17:
Next idea:
Check the groupId which the bulb listens on with MQTT command:
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgedevicefriendly_nameget_group_membership

IVI053, 2020-04-19:
> In my opinion the handling of the dimmer changed with the new firmware in the same way as for the remote https://www.zigbee2mqtt.io/information/binding.html#binding-a-remote-to-a-group.
>
> This [...]

I can fully confirm this and tested it with around 10 devices. Works

closed 2020-06-28 by stale[bot]

b4dpxl, 2020-07-10:
Thanks for this, I was having this problem before finding this issue. The docs should be update to reflect this.