Koenkk/zigbee2mqtt #3373: Z2M no longer starting

derekoharrow, 2020-04-16:
Z2M is no longer starting for me.

I'm running a CC2531 on a Raspberry Pi 3, which I have fully updated. I've even tried completely removing and re-installing Z2M from scratch. This was all working fine before yesterday.

When I do an npm start I get:

```
> zigbee2mqtt@1.12.2 start /opt/zigbee2mqtt
> node index.js

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-16T10_17_58_324Z-debug.log
```

The debug log says:
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.4
3 info using node@v12.16.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.12.2~prestart: zigbee2mqtt@1.12.2
6 info lifecycle zigbee2mqtt@1.12.2~start: zigbee2mqtt@1.12.2
7 verbose lifecycle zigbee2mqtt@1.12.2~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.12.2~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.12.2~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.12.2~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.12.2~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.12.2~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.12.2 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:310:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:310:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.12.2
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.97-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.16.2
19 verbose npm  v6.14.4
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.12.2 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.12.2 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
`

I've followed all the steps in the FAQ around failing to start, and everything checks out ok there.

Help!

Westcott1, 2020-04-16:
Sorry, but this is a "Me too!"
A couple of weeks ago, z2m wouldn't start anymore using _systemctl_, but would still start with 'npm start'. Now even the does not work.
When _zigbee-herdsman_ tries to start, the CC2531's green led goes out.
The port checks out OK, and I've tried reflashing.

**DEBUG=zigbee-herdsman* npm start**

> zigbee2mqtt@1.12.2-dev start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-04-16 20:32:53: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-04-16.20-32-53' filename: log.txt
zigbee2mqtt:debug 2020-04-16 20:32:54: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-04-16 20:32:54: Starting zigbee2mqtt version 1.12.2-dev (commit #29762b2)
zigbee2mqtt:info  2020-04-16 20:32:54: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-04-16 20:32:54: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"},"adapter":{"concurrent":null}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM1' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM1 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +8ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,0,2,6,3,194] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,65,128,0,2,0,2,6,3,194] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,0,2,6,3] - 194 +2ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":0,"majorrel":2,"minorrel":6,"hwrev":3} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +12ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,0,2,6,3,144,21,52,1,2,0,0,0,0,218] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,0,2,6,3,144,21,52,1,2,0,0,0,0] - 218 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608} +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}) +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +11ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,8,2,0,105] +25ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,8,2,0,105] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 8 - [2,0] - 105 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":2,"len":0,"value":{"type":"Buffer","data":[]}} +28ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
zigbee2mqtt:error 2020-04-16 20:32:55: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-04-16 20:32:55: Failed to start zigbee
zigbee2mqtt:error 2020-04-16 20:32:55: Exiting...
zigbee2mqtt:error 2020-04-16 20:32:55: Error: SREQ '--> SYS - osalNvRead - {"id":3840,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}' failed with status '2' (expected '0')
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.2-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.12.2-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-04-16T19_32_55_682Z-debug.log

Koenkk, 2020-04-16:
@Westcott1 you issue should be fixed in the latest dev branch

@derekoharrow I'm not sure about yours as I don't see any logging, if you reflashed your adapter and after this got the error it should also be fixed in the latest dev branch.

Westcott1, 2020-04-16:
Koenkk, thanks! All OK now

derekoharrow, 2020-04-17:
Well, this is embarassing!
Looks like it was the USB lead for the power. The PSU was fine, but the USB lead was causing low power. I've swapped it out and all seems ok now! Thanks.

closed 2020-04-17 by derekoharrow

Koenkk, 2020-04-17:
@derekoharrow could you give some more context on what you did, what fixed it and how you discovered this was the issue? (I'm asking because more people have this issue so it can be helpful for them).

derekoharrow, 2020-04-17:
Sure.

Basically, I couldn't figure out what was going wrong, but then I stumbled on some kernel messages using the dmesg command (I used dmesg -T -w to monitor the log).

First, I noticed some messages indicating issues talking to the cc2531 USB stick:
`Apr 17 10:16:01 domoticz kernel: [  527.917984] cdc_acm 1-1.5:1.0: failed to set dtr/rts`

I also noticed that the USB device was disconnecting and re-connecting:
```
Apr 17 10:00:58 domoticz kernel: [44929.156957] usb 1-1.5: USB disconnect, device number 119
Apr 17 10:00:59 domoticz kernel: [44929.455555] usb 1-1.5: new full-speed USB device number 120 using dwc_otg
Apr 17 10:00:59 domoticz kernel: [44929.604582] usb 1-1.5: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
Apr 17 10:00:59 domoticz kernel: [44929.604596] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 17 10:00:59 domoticz kernel: [44929.604606] usb 1-1.5: Product: TI CC2531 USB CDC
Apr 17 10:00:59 domoticz kernel: [44929.604615] usb 1-1.5: Manufacturer: Texas Instruments
Apr 17 10:00:59 domoticz kernel: [44929.604625] usb 1-1.5: SerialNumber: __0X00124B000E0EAA6F
Apr 17 10:00:59 domoticz kernel: [44929.607108] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device
```

The I noticed that this corresponded with under-voltage messages:
```
Apr 17 10:00:00 domoticz kernel: [44870.355590] Under-voltage detected! (0x00050005)
Apr 17 10:00:04 domoticz kernel: [44874.515618] Voltage normalised (0x00000000)
Apr 17 10:00:10 domoticz kernel: [44880.755718] Under-voltage detected! (0x00050005)
Apr 17 10:00:18 domoticz kernel: [44889.075627] Voltage normalised (0x00000000)
```

The only thing that had changed lately with my setup was that I had moved the Raspberry Pi and in doing so changed the PSU. I've ordered a new official Raspberry Pi PSU, but in the meantime thought that I'd at least try a different cable. Instantly, all these problems have just gone away!

Koenkk, 2020-04-17:
Thanks

> Apr 17 10:16:01 domoticz kernel: [ 527.917984] cdc_acm 1-1.5:1.0: failed to set dtr/rts

I also see this, I expect it's normal

> I also noticed that the USB device was disconnecting and re-connecting:

> The I noticed that this corresponded with under-voltage messages:

This of course not

I've added these tips to the FAQ https://www.zigbee2mqtt.io/information/FAQ.html#raspberry-pi-users-use-a-good-power-supply , thanks!