Koenkk/zigbee2mqtt #3374: Friendly name and MQTT discovery

silvanverschuur, 2020-04-16:
I'm not sure if this is a bug. I've added a `friendly_name` for all my devices so that Home Assistant is using this friendly name instead of the `unique_id`.

Until now every device that I own (Xiaomi smoke/temperature/buttons, Ikea power outlets/signal repeater, etc) is using this friendly name for MQTT discovery.

Recently I've added some Ikea bulbs and they get a different name in the MQTT discovery. They get a `_light` postfix. 

Example:
```
'0xccccccfffe709b7e':
  friendly_name: ikea_filament_bedroom

```
This becomes `light.ikea_filament_bedroom_light` in Home Assistant while other devices keep their original name (as specified in the Zigbee2Mqtt configuration file).

Is there any reason why a light gets the `_light` postfix and other devices keep their original friendly name?

## Debug Info
Zigbee2mqtt version: 1.12.0
Adapter hardware: CC2530 / CC26X2R1

Koenkk, 2020-04-16:
This is expected, the reason for this is e.g. a `sensor` device. Lets say this sensor (`my_sensor`) exposes both temperature and occupancy, without the postfix we would end up in duplicate entity. But when we add the type not: `sensor.my_sensor_temperature`, `sensor.my_sensor_occupancy`. For consistency purposes this has been done everywhere.

silvanverschuur, 2020-04-17:
Thanks for your explanation. Now I understand the reason behind it. Keep up doing the good work, I love z2m!

closed 2020-04-17 by silvanverschuur