Koenkk/zigbee2mqtt #3375: LLKZMK11LM bugs

jd1900, 2020-04-16:
# Bug Report
## What happened
I just added a LLKZMK11LM, but I have three different problems:
1.- Temperature is always 0. 
2.- Consumption is always 0 (power works).
3.- S1 and S2 are in reverse (see diagram https://ae01.alicdn.com/kf/HTB1h0CnRr2pK1RjSZFsq6yNlXXag.jpg). When I press the S1 switch, the l2 toggles, and viceversa with S2, the l1 toggles. I started just with L1 and S1, and it didn't change anything in the device, but I could see the S2 switch toggle in HA. So they seem to be totally wrong. I don't know wether this is a hardware problem or a software problem from zigbee2mqtt.

## What did you expect to happen
1.- Show Temperature.
2.- Show Consumption.
3.- Toggle the proper switch, S1 for L1 and S2 for L2.

## How to reproduce it (minimal and precise)
1.- Try to see temperature.
2.-Try to see Consumption. 
3.- Just connect L1 and S1, and see that the switch does nothing, but when connecting to S2 it changes the L1.

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20191106.zip

jd1900, 2020-04-16:
I've found another bug related to states consistency.

Here we have the info from HA:
[image]

If I set the state of the relay to on (z`igbee2mqtt/relay_aqara_2/set` --> `on`), both "State" and "State l1" switch to "On". Fine.
[image]

However, if I set the l1 state back to off (`zigbee2mqtt/relay_aqara_2/l1/set` --> `off`), general "State" doesn't change, only the L1 state, being the general "State" wrong:
[image]

jd1900, 2020-04-17:
I've found another bug.

ping is unsupported. When setting the availability_timeout <> 0 is does not change at all despite removing from the installation. In other devices it works (like Ikea bulbs).

Koenkk, 2020-04-17:
I've ordered a LLKZMK11LM (as I'm also interested in this device) so I will be able to check what's going with bug 1 2 and 3.

Regarding `zigbee2mqtt/relay_aqara_2/l1/set` the `state` not turning to off when controlling it without endpoint name (`l1`/`l2`). It's recommended to only control this device through the topic with endpoint in it.

Not sure where the color temp stuff comes from. I recommended clearing the `data/state.json` and see if it comes back.

jd1900, 2020-04-17:
Great. If you need any kind of testing/logs, let me know.

Regarding a hardware error, I discarded it. I took a second LLKZMK11LM and the errors are the same.

Fabiancrg, 2020-04-19:
I have some relays I am using for roller shutters and one I used to check points 2 and 3.

1. For me consumption looks OK, I am not really using this but I have some values, I think the reported value is in hWh with a precision of 10 Wh so maybe the load was not up long enough to see something ? With a load of 1 kW, you should see something after one minute.

2. I confirm that S1 is linked to L2 and S2 to L1.

3. Group look fine for me, I created a group 'Office' and added the relay(s) using the topic `zigbee2mqtt/bridge/group/Office/add` and message `Aqara_Relay/l1` or `Aqara_Relay/l2` or both
After that I binded a remote and was able to control L1, L2 or both, depensing what I added in the group.

For info, the topic I am using to control the relays is `zigbee2mqtt/Aqara_Relay/set` 
with message {"state_l1":"ON|OFF"} or {"state_l2":"ON|OFF"}

jd1900, 2020-04-19:
> 2. I confirm that S1 is linked to L2 and S2 to L1.

I wonder... If that is a hardware thing or software. If that is hardware, what a big failure..., if that's a software thing from Zigbee2Mqtt, then a lot of possibilities arise to control the relay.

> 3. Group look fine for me, I created a group 'Office' and added the relay(s) using the topic `zigbee2mqtt/bridge/group/Office/add` and message `Aqara_Relay/l1` or `Aqara_Relay/l2` or both
>    After that I binded a remote and was able to control L1, L2 or both, depensing what I added in the group.
Weird..., I'll try to reconfigure to see if it gets fixed.

 
> For info, the topic I am using to control the relays is `zigbee2mqtt/Aqara_Relay/set`
> with message {"state_l1":"ON|OFF"} or {"state_l2":"ON|OFF"}
Using direct messages I have no problem, just with groups/binding.

Fabiancrg, 2020-04-19:
I did more testing with groups and in fact it seems it's working until I send a message to one of the relays. After that, only L2 will continue to be manage by the remote binded to the group.
Maybe it's not supporting very well the 'toggle' sent by the remote, I will try with another switch tomorrow as controlling the group directly does not seem to be working.

Fabiancrg, 2020-04-20:
@jd1900 some updates
1. I confirm consumption is reported correctly with a precision of 10 Wh.
2. Don't know if this is something embeded in the firmware or in Z2M, I suppose @Koenkk will be able to answer to this. But it's not really un issue if it is documented, I can create a PR to document it if it is confirmed it is a FW bug.
3. I tried with another switch (IKEA E1743) and I can bind it to a group containing L2 but not to the group with L1. I cannot bind the switch directly to any end point but it is normal as it isn't supported by this switch.

jd1900, 2020-04-20:
> @jd1900 some updates
> 
> 1. I confirm consumption is reported correctly with a precision of 10 Wh.

Maybe my bulb LEDs i've connected consume too little... I have no other explanation... 

> 2. Don't know if this is something embeded in the firmware or in Z2M, I suppose @Koenkk will be able to answer to this. But it's not really un issue if it is documented, I can create a PR to document it if it is confirmed it is a FW bug.

I've search through the internet and have found nothing about it. If that only happens in Z2M would be very weird. One thing to try would be to test before pairing, to see how it behaves.

> 3. I tried with another switch (IKEA E1743) and I can bind it to a group containing L2 but not to the group with L1. I cannot bind the switch directly to any end point but it is normal as it isn't supported by this switch.

Actually you can bind to the group containing L1, but L1 doesn't respond. Actually, I have both L1 and L2 in a common group, and using a bound device or directly through MQTT there is no answer to the group from L1.

So you basically confirm that I can't bind the Aqara switch to a group... it's a pity, since it would solve my biggest problem.

Fabiancrg, 2020-04-20:
@jd1900 I checked on an out of the box module, did not pair it to anything and connected the switches. It seems the bug is in the module firmware as S1 is linked to L2 and S2 to L1.

airborn, 2020-05-24:
The actual problem with S1 and S2 may by a lot more complicated then just switched labels. Take look at [this thread](https://community.hubitat.com/t/new-xiaomi-aqara-relay-2-channel/13335/91) I have very similar behavior with my units. The S1 and S2 can actualy switch both outputs in a non deterministic way. Everything works fine when switching this from HA, but S1 and S2 appears to be totally screwed.

closed 2020-07-26 by stale[bot]

cody82, 2020-09-03:
My experience with that device:
I am using the LLKZMK11LM to control shutters. S1 and S2 behaviour looks ok, although I did not test it much.
But when controlling both relays with one message like {"state_l1":"ON","state_l2":"OFF"} it behaves totally random.
When switching on interlock and using {"state_l1":"ON"} instead it works.

leroyloren, 2021-03-15:
Hi, I control the heating ladder (600 W) via this relay, but the z2m displays strange values when measuring current, including consumption.

[image]

cody82, 2021-03-17:
Values for my shutters(100W) look ok.