Koenkk/zigbee2mqtt #3376: Gledopto GL-B-001ZS

kutik666, 2020-04-16:
Hello

Gledopto GL-B-001ZS pair, but dont show in mqtt

Lada

swissbyte, 2020-04-22:
Please enable debug mode for herdsman and post your log file here.

kutik666, 2020-04-22:
this log produce after change record in file device.js
GL-B-001Z to GL-B-001ZS

info  2020-04-17 20:32:50: Currently 6 devices are joined:
info  2020-04-17 20:32:50: 0x00158d0004074e07 (0x00158d0004074e07): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
info  2020-04-17 20:32:50: 0x00158d0003ccfcfd (0x00158d0003ccfcfd): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
info  2020-04-17 20:32:50: 0x00158d0004450704 (0x00158d0004450704): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2020-04-17 20:32:50: 0x04cf8cdf3c773ca7 (0x04cf8cdf3c773ca7): GZCGQ01LM - Xiaomi MiJia light intensity sensor (EndDevice)
info  2020-04-17 20:32:50: 0x00124b001f3205cb (0x00124b001f3205cb): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
info  2020-04-17 20:32:50: 0x00124b001ed005a4 (0x00124b001ed005a4): GL-B-001ZS - Gledopto Smart 4W E14 RGB / CCT LED bulb (Router)

 Device '0x00124b001ed005a4' announced itself
info  2020-04-17 20:33:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ed005a4"}}'

Device '0x00124b001ed005a4' announced itself
info  2020-04-17 20:43:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ed005a4"}}'
debug 2020-04-17 20:43:02: Received Zigbee message from '0x04cf8cdf3c773ca7', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":4772}' from endpoint 1 with groupID 0
info  2020-04-17 20:43:02: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c773ca7', payload '{"battery":100,"voltage":3100,"linkquality":60,"illuminance":4772,"illuminance_lux":2}'
debug 2020-04-17 20:43:06: Device '0x00124b001ed005a4' announced itself
info  2020-04-17 20:43:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ed005a4"}}'
debug 2020-04-17 20:43:11: Received Zigbee message from '0x00124b001f3205cb', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-04-17 20:43:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001f3205cb', payload '{"state":true,"led_state":false,"linkquality":60,"description":"8877/0x00158D0004074E07","type":"CHILD_RFD","rssi":108}'
debug 2020-04-17 20:43:11: Received Zigbee message from '0x00124b001f3205cb', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B001CD47AC0","inactiveText":"COORD","presentValue":17,"relinquishDefault":0,"minimumOffTime":0}' from endpoint 8 with groupID 0
info  2020-04-17 20:43:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001f3205cb', payload '{"state":true,"led_state":false,"linkquality":60,"description":"0/0x00124B001CD47AC0","type":"COORD","rssi":17}'
debug 2020-04-17 20:43:11: Received Zigbee message from '0x00124b001f3205cb', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"8877/0x00158D0004074E07","inactiveText":"CHILD_RFD","presentValue":119,"relinquishDefault":2,"minimumOffTime":0}' from endpoint 8 with groupID 0
info  2020-04-17 20:43:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001f3205cb', payload '{"state":true,"led_state":false,"linkquality":60,"description":"8877/0x00158D0004074E07","type":"CHILD_RFD","rssi":119}'
debug 2020-04-17 20:43:11: Device '0x00124b001ed005a4' announced itself
info  2020-04-17 20:43:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ed005a4"}}'
debug 2020-04-17 20:43:17: Device '0x00124b001ed005a4' announced itself
info  2020-04-17 20:43:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ed005a4"}}'
warn  2020-04-17 20:43:22: Device '0x00124b001ed005a4' left the network
info  2020-04-17 20:43:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00124b001ed005a4"}}'
info  2020-04-17 20:43:30: Device '0x00124b001ed005a4' joined
info  2020-04-17 20:43:30: Starting interview of '0x00124b001ed005a4'
info  2020-04-17 20:43:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001ed005a4"}}'
info  2020-04-17 20:43:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001ed005a4"}}'
debug 2020-04-17 20:43:30: Device '0x00124b001ed005a4' announced itself
info  2020-04-17 20:43:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ed005a4"}}'
debug 2020-04-17 20:43:30: Device '0x00124b001ed005a4' announced itself
info  2020-04-17 20:43:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b001ed005a4"}}'
debug 2020-04-17 20:43:32: Received Zigbee message from '0x04cf8cdf3c773ca7', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":6021}' from endpoint 1 with groupID 0
[... cut ...] rssi":37}'
debug 2020-04-17 20:45:11: Received Zigbee message from '0x00124b001f3205cb', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B001CD47AC0","inactiveText":"COORD","presentValue":24,"relinquishDefault":0,"minimumOffTime":0}' from endpoint 8 with groupID 0
info  2020-04-17 20:45:11: MQTT publish: topic 'zigbee2mqtt/0x00124b001f3205cb', payload '{"state":true,"led_state":false,"linkquality":68,"description":"0/0x00124B001CD47AC0","type":"COORD","rssi":24}'

elektrikal80, 2020-05-01:
I have the same issue with this lamp Gledopto GL-B-001ZS

kutik666, 2020-05-24:
Work OK in  1.13.1.1
thanks

Peerkersezuuker, 2020-07-22:
I also have this issue.
I am running v.1.14.1-dev (zStack12 20190619) in combo with domoticz.
The device is paired, and can be seen in the database and in the zigbee2mqtt plugin in domoticz, but no devices are created.
The stick is running CC2531_SOURCE_ROUTING_20190619

closed 2020-08-30 by github-actions[bot]