Koenkk/zigbee2mqtt #3379: Error and missing battery information Smartthings 3321-S Multipurpose Sensor

whatdaybob, 2020-04-16:
# Bug Report
## What happened

Error in log saying `Failed to interview 'SmartThings Multi Sensor (2015 model)', device has not successfully been paired` but device seems to work. I have removed and added the device multiple times and it does the same. Also the exact battery is not reported, it is using a battery_low boolean but I believe battery percentage is available in other implementations. 

## What did you expect to happen

No errors to be thrown. Battery percentage to be reported,

## How to reproduce it (minimal and precise)

Add device to zigbee2mqtt, wait a minute or two after device is reporting

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version: Flashed with CC2531_DEFAULT_20190608.zip

Simpler Error log available [Hastebin](https://hastebin.com/rajeciwaju.coffeescript)
Larger Error log with interview attempt [Hastebin](https://hastebin.com/unucerenah.makefile)

Koenkk, 2020-04-17:
I assume the interview succeeded at least once, otherwise it wouldn't  able to identify it. Is the larger log of the non-first attempt?

Can you remove the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove and try to pair it again close to the coordinator?

whatdaybob, 2020-04-17:
You assume correctly that it succeeded. The larger log was just a few minutes after the initial pairing. The attempt these logs show have already done as you have suggested. Happy to do it again if you need me to dump the entire log from the docker start to me pairing to it erroring but there isn't much more produced.

closed 2020-06-26 by stale[bot]