Koenkk/zigbee2mqtt #3380: Schneider Button Zigbee : Harmony ZBRT1

NicoFouch, 2020-04-17:
Hello,
I would like to use a Zigbee Schneider button with my zigbee2mqtt. I already connected two zigbee aqara xiaomi and they are very efficient.
I will want help to connect the Zigbee Schneider Harmony ZBRT1.
It's the SmartRF Packet Sniffer at the bottom.
[image]
When I push the button, there are 4 frame to send. But when I want to associate my button with the Zigbee2MQTT gateway then there is no return.
Best regards.

swissbyte, 2020-04-22:
Hi NicoFouch

Its nice to see that you want to expand the supported devices list of z2m. 
Please try to connect your Schneider button to z2m and post the log-file of the herdsman here. Then we will have the relevant information to start. 

Here is a guide: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

If you have further questions, feel free to ask.

ryanbeaton, 2020-04-22:
That one might be hard work as its green power.  From https://www.se.com/ww/en/product/ZBRT1/transmitter-wireless-batteryless-w-o-head/

communication port protocol: Zigbee green power at 2.4 GHz conforming to IEEE 802.15.4

sjorge, 2020-04-22:
There is Zigbee GP support in the latest dev version though, so it might already just works as there aren't many models like for normal zigbee devices.

NicoFouch, 2020-04-23:
Hello,
thank you to help me.
I can not appear my ZBRT1. There is no matching mode. I can't send you the log-file of the herdsman.

NicoFouch, 2020-05-11:
Hello,

If there is someone who is ready to help me, I can send you a ZBRT1 Schneider button by mail so you can do it. Thank you very much for helping out.

closed 2020-07-18 by stale[bot]