Koenkk/zigbee2mqtt #3381: Issue starting new Zigbee2MQTT docker instance.

eyeson, 2020-04-17:
Trying to move my zigbee2mqtt install form a raspberry pi into my VM. 
I have spun a docker compose file with the following settings:
`  zigbee2mqtt:
    image: koenkk/zigbee2mqtt
    container_name: zigbee2mqtt
    environment:
      - TZ=Europe/London
    volumes:
      - /home/uvm/Config/Z2M:/app/data
      - /run/udev:/run/udev:ro
    network_mode: host
    privileged: true
    restart: unless-stopped
`
I've included a device and removed it, both get the same error, zigbee herdsman doesn't want to start.

`zigbee2mqtt      | Using '/app/data' as data directory
zigbee2mqtt      | 
zigbee2mqtt      | > zigbee2mqtt@1.12.2 start /app
zigbee2mqtt      | > node index.js
zigbee2mqtt      | 
zigbee2mqtt      | zigbee2mqtt:info  2020-04-17 17:15:21: Logging to console and directory: '/app/data/log/2020-04-17.17-15-21' filename: log.txt
zigbee2mqtt      | zigbee2mqtt:info  2020-04-17 17:15:21: Starting zigbee2mqtt version 1.12.2 (commit #7d27a54)
zigbee2mqtt      | zigbee2mqtt:info  2020-04-17 17:15:21: Starting zigbee-herdsman...
zigbee2mqtt      | zigbee2mqtt:error 2020-04-17 17:16:22: Error while starting zigbee-herdsman
zigbee2mqtt      | zigbee2mqtt:error 2020-04-17 17:16:22: Failed to start zigbee
zigbee2mqtt      | zigbee2mqtt:error 2020-04-17 17:16:22: Exiting...
zigbee2mqtt      | zigbee2mqtt:error 2020-04-17 17:16:22: Error: AREQ - ZDO - stateChangeInd after 60000ms
zigbee2mqtt      |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
zigbee2mqtt      |     at listOnTimeout (internal/timers.js:549:17)
zigbee2mqtt      |     at processTimers (internal/timers.js:492:7)
zigbee2mqtt      | npm ERR! code ELIFECYCLE
zigbee2mqtt      | npm ERR! errno 1
zigbee2mqtt      | npm ERR! zigbee2mqtt@1.12.2 start: `node index.js`
zigbee2mqtt      | npm ERR! Exit status 1
zigbee2mqtt      | npm ERR! 
zigbee2mqtt      | npm ERR! Failed at the zigbee2mqtt@1.12.2 start script.
zigbee2mqtt      | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
zigbee2mqtt      | 
zigbee2mqtt      | npm ERR! A complete log of this run can be found in:
zigbee2mqtt      | npm ERR!     /root/.npm/_logs/2020-04-17T16_16_22_953Z-debug.log
uvm@uvm-BHYVE:~$ 

`

Anything else I can try?

Koenkk, 2020-04-17:
Make sure to copy over all the contents of the `data` directory, especially `data/coordinator_backup.json` is important (assuming your are using  a CC26x2R1 or CC1352-P2)

eyeson, 2020-04-17:
Perfect - I did that and it now works! Cheers

closed 2020-04-17 by eyeson

NicolasBoulanger, 2020-05-26:
hello
is it mandatory tu add advanced configuration like in hassio addon, in data/configuration.yaml docker?  (like network key etc..
which configuration take the container by default with no advanced config?
thanks