Koenkk/zigbee2mqtt #3382: Request: LLKZMK11LM binding

jd1900, 2020-04-17:
In the LLKZMK11LM (https://www.zigbee2mqtt.io/devices/LLKZMK11LM.html) relay would be great to bind the toggling/switching signal to a device, so that when the switch is pressed it can send an OnOff action to a bound device (like a bulb, that could be, or not, attached to it).

Koenkk, 2020-04-17:
I'm not sure if this device support it. You can check if it does by trying https://www.zigbee2mqtt.io/information/binding.html . Note that such feature has to be supported by the device itself, it's not something that can be added from zigbee2mqtt.

jd1900, 2020-04-17:
I tried, but I got this response:
zigbee2mqtt:error 2020-04-17 21:53:07: Nothing to bind from 'relay_aqara_1' to 'light_group'

Koenkk, 2020-04-18:
Cam you share your database.db?

jd1900, 2020-04-18:
Sure. 
How can I share just to you? To avoid leaving it public.

Koenkk, 2020-04-19:
Sharing the entry of `LLKZMK11LM` in database.db is enough, it doesn't contain any sensitive information (maybe the ieeeAddr but you can remove that).

jd1900, 2020-04-19:
There you go. From two of them. I've replaced the last bytes of the Addresses.
I'm intrigued why they differ in important things such as inClusterList or be powered by battery (maybe because I disconnected a few days ago one of them???).

{"id":9,"type":"Group","groupID":3,"members":[{"deviceIeeeAddr":"0x680aanotherdevice","endpointID":1},{"deviceIeeeAddr":"0x00158d000replaced","endpointID":2},{"deviceIeeeAddr":"0x00158d000replaced","endpointID":1},{"deviceIeeeAddr":"0x00158d000replaced2","endpointID":2},{"deviceIeeeAddr":"0x00158d000replaced2","endpointID":1}],"meta":{}}

{"id":10,"type":"Router","ieeeAddr":"0x00158d000replaced","nwkAddr":29481,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.relay.c2acn01","epList":[1,2],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"65281":{"3":41,"5":9,"7":"0x0000000000000000","8":4644,"9":1026,"100":0,"101":0,"110":0,"111":0,"148":4,"149":0.0021875000093132257,"150":2308.025390625,"151":0.005009882152080536,"152":0.051727294921875,"155":0},"modelId":"lumi.relay.c2acn01","appVersion":1}},"genOnOff":{"attributes":{"61440":117440568,"onOff":0}},"genAnalogInput":{"attributes":{"presentValue":0.01050710678100586}}},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":124749678,"onOff":0}}},"binds":[]}},"appVersion":1,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1587037165939}

{"id":11,"type":"Router","ieeeAddr":"0x00158d000replaced2","nwkAddr":42437,"manufId":4151,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.relay.c2acn01","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,1,2,10,6,16,2820,12],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":0,"5":9,"7":"0x0000000000000000","8":4644,"9":769,"100":0,"101":0,"110":0,"111":0,"148":6,"149":0.0003124999930150807,"150":2317.1650390625,"151":0.0035429000854492188,"152":0.051727294921875,"155":0},"modelId":"lumi.relay.c2acn01","appVersion":1,"manufacturerName":"LUMI","powerSource":4,"zclVersion":0,"stackVersion":2,"hwVersion":18,"dateCode":"02-27-2019"}},"genOnOff":{"attributes":{"61440":117440579,"onOff":0}},"genAnalogInput":{"attributes":{"presentValue":0.15316128730773926}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa6b4d","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[6,16,4,5],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":117440578,"onOff":0}}},"binds":[]}},"appVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"02-27-2019","zclVersion":0,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1587050344626}

Koenkk, 2020-04-20:
The device does not have any outgoing clusters (`outClusterList":[]`) meaning it does not support controlling other devices.

jd1900, 2020-04-20:
Thank you. Thus, close the issue.

jd1900, 2020-04-20:
Sorry to reopen, why do they have those differences between the two LLKZMK11LM?

Koenkk, 2020-04-20:
@jd1900 because Xiaomi devices are hard to interview so there is some fallback mechanism to make interview succeed with some assumptions. However as even the fully interviewed one has an empty out cluster list it doesn't support it.

In case you want to fix the entry it self you can try repairing it by first removing it: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

jd1900, 2020-04-20:
Thank you. 
Time to close now.

closed 2020-04-20 by jd1900