Koenkk/zigbee2mqtt #3383: Request: Disassociate relay/switch message from on/off functions in LLKZMK11LM

jd1900, 2020-04-17:
I was wondering if this could be possible. For any relay, and specifically for this (LLKZMK11LM).

Most relays/switches have automatic on/off functions when the switch is activated.

Pressing the physical switch gives the signal to the relay to change the on/off.

But..., is it possible to disassociate both functions?
1.- Pressing the physical switch to just send a signal / mqtt message, but not toggling on/off.
2.- (already possible) Turning on/off just by message.

What could this be for?

If we have a smart bulb with a physical switch with the smart relay, pressing the switch turns off the smart bulb by cutting the power off, instead of sending the signal to the bulb to turn off keeping it powered. We could use an automation to turn/off the smart bulb, or even better, binding  (request related https://github.com/Koenkk/zigbee2mqtt/issues/3382) the signal instead of creating an automation and not depending on HA.

Koenkk, 2020-04-17:
Such feature has to be supported by the firmware on the LLKZMK11LM (which I'm not of sure) and is not something that can be added by zigbee2mqtt.

jd1900, 2020-04-17:
I just asked because I didn't know if it was added or done by firmware. In that case, I close the case. 
Thank you.

closed 2020-04-17 by jd1900