Koenkk/zigbee2mqtt #3384: How can set new Parameter on Spirit Zigbee with zigbee2mqtt and FHEM?

ClashC, 2020-04-18:
Hello all,

Since yesterday I use my new Eurotronics Spirit ZigBee with CC2531 ZigBee USB stick. The USB stick is flashed with the latest firmware.
I have now checked out and installed zigbee2mqtt on one from yesterday on a raspi3b+, all without errors. 
This page has helped me: http://wiki.gorjup.de/doku.php?id=public:fhem_zigbee_cc2531#fhem_zigbee_cc2531_installation_und_konfiguration_auf_einem_raspberry_pi

Now, after learning the Spirit ZigBee, I see it as a new device at FHEM. I get a lot of readings from Spirit ZigBee...
But how can I set a new temperature from FHEM?
How can I set the boost mode?
Does anyone have a tip?

TheDodger, 2020-04-18:
Here's my FHEM setList for these devices:

`system_mode:off,auto,heat zigbee2mqtt/kidsroom_heat/set {"$EVTPART0":"$EVTPART1"}
current_heating_setpoint:selectnumbers,5,0.5,30,1,lin zigbee2mqtt/kidsroom_heat/set {"$EVTPART0":"$EVTPART1"}
occupied_heating_setpoint:selectnumbers,5,0.5,30,1,lin zigbee2mqtt/kidsroom_heat/set {"$EVTPART0":"$EVTPART1"}
unoccupied_heating_setpoint:selectnumbers,5,0.5,30,1,lin zigbee2mqtt/kidsroom_heat/set {"$EVTPART0":"$EVTPART1"}
eurotronic_system_mode:textField zigbee2mqtt/kidsroom_heat/set {"$EVTPART0":"$EVTPART1"}
local_temperature_calibration:selectnumbers,-2.5,0.5,2.5,1,lin zigbee2mqtt/kidsroom_heat/set {"$EVTPART0":"$EVTPART1"}
remote_sensing:textField zigbee2mqtt/kidsroom_heat/set {"$EVTPART0":"$EVTPART1"}`

Just experiment a bit and watch the device readings. There are some limitations with setting temperatures, `eurotronic_system_mode` is a bit field that's documented in the manual of the device and `remote_sensing` does basically nothing.

Don't forget to close this issue.

ClashC, 2020-04-18:
Hello TheDodger,

actually I know fhem quite well, but I don't see where and how I can put your info into my setup. Could you give me a little example?

With a mqtt client I was able to integrate the Spirit with the info from here: https://www.zigbee2mqtt.io/devices/SPZB0001.html
well change the settings. 
I would like to do this only via FHEM and the created device...

> define xBridge XiaomiMQTTDevice bridge
> setuuid xBridge 5e99cb2d-f33f-cb17-3863-3b69641778bfb3b6
> attr xBridge IODev myBroker2
> attr xBridge room MQTT,XiaomiMQTTDevice
> define Coordinator XiaomiMQTTDevice unknown 0x00124b0018e2ebb6 Coordinator zigbee2mqtt
> setuuid Coordinator 5e99cb2f-f33f-cb17-4699-9c6c64a11e7bd924
> attr Coordinator IODev myBroker2
> attr Coordinator room XiaomiMQTTDevice
> attr Coordinator stateFormat state
> define FileLog_Coordinator FileLog ./log/Coordinator-%Y-%m.log Coordinator
> setuuid FileLog_Coordinator 5e99cb2f-f33f-cb17-3ae3-c6bc9be5ec638346
> attr FileLog_Coordinator logtype text
> attr FileLog_Coordinator room XiaomiMQTTDevice
> define FileLog_SpiritZigbee FileLog ./log/SpiritZigbee-%Y-%m.log SpiritZigbee
> setuuid FileLog_SpiritZigbee 5e99fcc7-f33f-cb17-9855-5017869063c13796
> attr FileLog_SpiritZigbee alias SpiritZigbee
> attr FileLog_SpiritZigbee logtype text
> attr FileLog_SpiritZigbee room XiaomiMQTTDevice
> define SpiritZigbee XiaomiMQTTDevice SPZB0001 0x00158d0003c4726e SpiritZigbee zigbee2mqtt
> setuuid SpiritZigbee 5e9a0029-f33f-cb17-b608-06309bbee65d9edb
> attr SpiritZigbee IODev myBroker2
> attr SpiritZigbee event-min-interval .*:200
> attr SpiritZigbee room Bad,MQTT,Temperaturen,XiaomiMQTTDevice
> attr SpiritZigbee stateFormat Soll: occupied_heating_setpoint °C --- Ist: local_temperature °C

closed 2020-06-26 by stale[bot]