Koenkk/zigbee2mqtt #3388: CC1352R (not P) adapter support

nattgris, 2020-04-18:
Hi!

I'm planning to swap my CC2531 coordinator for a CC26X2R1 or CC1352P_2. However I don't like the look of a TI devboard dangling in a cable from my otherwise sleek looking z2m RPi. I've been looking around for a compatible module that I can build a RPi HAT for. Unfortunately it appears to be hard to find a module with one of the two supported chips. I did find a CC1352R module that could work, but before I give it a try, I'd like to know if this variant can be added to the list of supported adapters.

I would assume that the CC2652R and CC1352R are similar enough that this would not be difficult. Maybe they are even binary compatible, as long as nothing touches the sub-GHz parts? Perhaps even the R and P versions can be auto-detected at runtime by the same FW and the PA configured only if available.

Is there anything else on the TI devboards that the ZNP firmware depends on that I would have to implement, or will it be enough to keep the UART (maybe LEDs also) on the same pins?

Another question is whether it actually makes any sense to use the PA versions of the chips in a Zigbee coordinator. Increasing the output power of one peer generally doesn't provide better range since your still limited by the link-budget of the reverse path. But maybe Zigbee is smart enough to handle multi-hop return paths even if the forward path can be done in a single hop?

Koenkk, 2020-04-18:
- This chip requires a different firmware however compiling it shouldn't be very difficult. You can follow the instructions from here: https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/COMPILE.md. For step 2 you need to import the CC1352R stack.
- Perhaps you are interested in the ZZH (https://electrolama.com/projects/zig-a-zig-ah/), not available yet but hopefully soon.

nattgris, 2020-06-17:
Thanks, this answers my question.

closed 2020-06-17 by nattgris