Koenkk/zigbee2mqtt #3391: docker koenkk/zigbee2mqtt:latest-dev no longer able to connect to MQTT server

tedsluis, 2020-04-19:
# Bug Report
## What happened
I run docker koenkk/zigbee2mqtt:latest-dev on intel NUC I5 core for quite some time. I update the docker images every day. Since a couple of days it can no longer connect to MQTT. I get messages like:
```
zigbee2mqtt:error 2020-04-19 08:42:08: Not connected to MQTT server!
zigbee2mqtt:error 2020-04-19 08:42:08: Cannot send message: topic: 'homeassistant/binary_sensor/0x00124b001bb859ff/router/config', .......
```
Running koenkk/zigbee2mqtt:latest still works!

## What did you expect to happen
I expect messages like:
```
zigbee2mqtt:info  2020-04-19 08:36:12: MQTT publish: topic 'homeassistant/sensor/0x00158d00027257cd/temperature/config', payload .........
```
I also expect to see messages reporting zigbee2mqtt to connect to MQTT in the logging after a restart, like this:
```
zigbee2mqtt:info  2020-04-19 08:36:12: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-04-19 08:36:12: Connecting to MQTT server at mqtt://192.168.11.100
zigbee2mqtt:info  2020-04-19 08:36:12: Connected to MQTT server
zigbee2mqtt:info  2020-04-19 08:36:12: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
```

## How to reproduce it (minimal and precise)
When I pulled a new version of the koenkk/zigbee2mqtt:latest-dev docker image some days ago, zigbee2mqqt started, but it failed to connect to the MQTT server, see [zigbee2mqtt-dev.log](https://pastebin.com/WA3CC9r9). 
MQTT server is running on the same dockerhost.
I also captured logging with debug level on: [zigbee2mqtt-dev-debug.log](https://pastebin.com/fC3VYb9R). Even the latest-dev image is still not able to connect to MQTT.

Also, when I execute the command below inside the container, it does not work, the light does not turn ON and I see nothing in my MQTT server log:
```
# docker exec zigbee2mqtt  /app/node_modules/.bin/mqtt publish -h MQTTHOSTNAME -u admin -P MQTTSERVERPASSWORD -v -t zigbee2mqtt/lampkeuken/set -m '{   "state": "ON", }'
```
See debug log from commandline: https://pastebin.com/5QhBQjqB

To fix it, I moved back to the  koenkk/zigbee2mqtt:latest docker image (stable verson). It immediately worked, zigbee2mqtt was able to connect to MQTT, see [zigbee2mqtt-stable.log](https://pastebin.com/FXccC9Ww).

Also, when I execute the command below inside the container, the light turn ON:
```
# docker exec zigbee2mqtt  /app/node_modules/.bin/mqtt publish -h MQTTHOSTNAME -u admin -P MQTTSERVERPASSWORD -v -t zigbee2mqtt/lampkeuken/set -m '{   "state": "ON", }'
```
I have been switching between the koenkk/zigbee2mqtt:latest docker image and koenkk/zigbee2mqtt:latest-dev docker image in the last few days, and the result is always the same: koenkk/zigbee2mqtt:latest-dev is not able to connect to the MQTT server.

## Debug Info
Zigbee2mqtt version: [koenkk/zigbee2mqtt:latest-dev](https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest-dev/images/sha256-dc343d5f4ba52dee7a59d04a6b0040579f4ae6efd8bbfb08fe30dbf7e9672682?context=explore)
Adapter hardware: CC1352P-2
Adapter firmware version: {'type': 'zStack3x0', 'meta': {'transportrev': 2, 'product': 1, 'majorrel': 2, 'minorrel': 7, 'maintrel': 1, 'revision': 20191106}}

My Docker file:
```
Description=Zigbee2mqtt
After=docker.service mqtt.service 
Requires=docker.service mqtt.service network.target 

[Service]
RestartSec=3
TimeoutStartSec=300
TimeoutSec=180
Restart=always
ExecStartPre=-/usr/bin/docker stop -t 2 zigbee2mqtt
ExecStartPre=-/usr/bin/docker rm -f zigbee2mqtt
ExecStartPre=/usr/bin/docker pull koenkk/zigbee2mqtt:latest-dev
ExecStart=/usr/bin/docker run --name="zigbee2mqtt" \
                              -e DEBUG=* \
                              -v /home/homeassistant/zigbee2mqtt:/app/data \
                              -e TZ=Europe/Amsterdam \
                              --privileged=true \
                              --memory="1000m" \
                              koenkk/zigbee2mqtt:latest-dev
ExecStop=-/usr/bin/docker stop -t 2 zigbee2mqtt
ExecStopPost=-/usr/bin/docker rm -f zigbee2mqtt

[Install]
WantedBy=multi-user.target
```
My configuration.yaml:
```
homeassistant: true                      
permit_join: false                       
mqtt:                   
  base_topic: zigbee2mqtt
  server: 'mqtt://*************'
  user: admin
  password: ************
  reject_unauthorized: false
  include_device_information: true
zigbee_shepherd_devices: false
serial:    
  port:    /dev/ttyACM0       
  #disable_led: true     
advanced:                   
  network_key: ***************
  pan_id: 0x165a
  ext_pan_id: **********
  channel: 26
  cache_state: true
  log_level: info
  last_seen: ISO_8601
  elapsed: false
  availability_timeout: 30
  report: true
  rtscts: true
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: hass/status
  timestamp_format: 'YYYY-MM-DD HH:mm:ss'
  adapter_concurrent: 16
map_options:
  graphviz:
    colors:
      fill:
        enddevice: '#fff8ce'
        coordinator: '#e04e5d'
        router: '#4ea3e0'
      font:
        coordinator: '#ffffff'
        router: '#ffffff'
        enddevice: '#000000'
      line:
        active: '#009900'
        inactive: '#994444'
#queue:
#  delay: 250
#  simultaneously: 5
device_options:
  retain: false
  qos: 1
  debounce: 1
  debounce_ignore:
  - action
  - brightness
  - occupancy
  - light
  - switch_l1
  - switch_l2
  - switch
  - contact
  - click
```

tedsluis, 2020-04-21:
I hoped it had something to do with pull request [Use CMD alongside ENTRYPOINT in the Dockerfile](https://github.com/Koenkk/zigbee2mqtt/pull/3366), but that one was reverted [Revert "Use CMD alongside ENTRYPOINT in the Dockerfile"](https://github.com/Koenkk/zigbee2mqtt/pull/3389) yesterday and zigbee2mqtt still can not connect to the MQTT server. 

Is there anything else I can check to troubleshoot this?

Koenkk, 2020-04-21:
I did not see any reports with MQTT connection not working in the latest dev.

- Try without `reject_unauthorized: false`.
- Check the logging of the MQTT server
- What MQTT server are you using?

tedsluis, 2020-04-22:
> Try without reject_unauthorized: false.
I removed this setting and that was the trick! Thanks @Koenkk !

closed 2020-04-22 by tedsluis