Koenkk/zigbee2mqtt #3392: Add a new device with a private Cluster ID

SillyDay, 2020-04-19:
# Bug Report
## What happened
  I have some ZigBee 3.0 modules(From EBYTE), and they use Private Cluster ID(0xFEB0) and Private Profile ID(0x0504). When I add it to the network (The module didn’t have modelID so I used manufacturerID instead.) and trigger an action, nothing happened. 
  So I captured the Serial communication between the Coordinator(CC2652R) and zigbee2mqtt, I found the Coordinator send nothing to the Serial Port when received the packet which Cluster ID and Private Profile ID is Private(Actually the Coordinator received and forwarded this packet in Zigbee network).I believe that's the key.
   So is there any ways to use Private Cluster ID and Private Profile ID? Should I modify the Coordinator firmware or some code in Zigbee2MQTT?
[image]
## What did you expect to happen
The data from Private Cluster ID and Private Profile ID can be readable
## How to reproduce it (minimal and precise)
Add a device using Private Cluster ID and Private Profile ID into the network.
## Debug Info
Zigbee2mqtt version: 1.12.2-dev
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200417

Koenkk, 2020-04-20:
To add a new cluster you have to add it here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts

SillyDay, 2020-04-20:
> To add a new cluster you have to add it here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts

Hi Koenkk,
  I added following lines into cluster.js

	manuSpecificEBYTEDevices: {
        ID: 0xFEB0,
		manufacturerCode: manufacturerCode_1.default.EBYTE,
        attributes: {
		// attributes seems to vary depending on the device. Can't be static
        },
        commands: {},
        commandsResponse: {},
    },

and added EBYTE: 0x1002, into the manufacturerCode.js. But it not works.

devices.js:

    {
        zigbeeModel: ['EBYTE'],
        model: 'EBYTE',
        vendor: 'EBYTE',
        description: 'ZigBee',
        supports: 'test',
        fromZigbee: [],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['manuSpecificEBYTEDevices']);
        },
    },

IN ADDITION，the device don't have the zigbeeModel indeed，so I modify the index.js like this:

function findByDevice(device) {

	if (device.manufacturerID === 4098){//EBYTE device
		console.log('EBYTE！');
		device.modelID = 'EBYTE';
	}

    let definition = findByZigbeeModel(device.modelID);

    if (!definition) {
        // Find by fingerprint
        loop:
        for (const definitionWithFingerprint of withFingerprint) {
            for (const fingerprint of definitionWithFingerprint.fingerprint) {
                if (fingerprintMatch(fingerprint, device)) {
                    definition = definitionWithFingerprint;
                    break loop;
                }
            }
        }
    }
    return definition;
}

But I haven't received the data so far.  I think I must be doing something wrong, pls correct me.
Thanks a lot!

SillyDay, 2020-04-20:
@Koenkk I think the key to this problem is that the coordinator is not listening to private profile ID 0x0504, so I added relevant content in the startZnp.js and deleted the database.db

    Object.assign(Object.assign({}, EndpointDefaults), { endpoint: 200, appprofid: 0x0504 }),

But it not works.

Koenkk, 2020-04-21:
@SillyDay do you see any data coming in when running with the herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

SillyDay, 2020-04-21:
> @SillyDay do you see any data coming in when running with the herdsman debug logging?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

@Koenkk I got nothing useful in herdsman logs. I think my modification is incorrect. In my case, the device used Profile ID(0x0504),and the Dest. Endpoint ID is 1,show as follows. How can I add this profile ID into herdsman correctly and which files do I need to modify?

Thanks!

[image]

Koenkk, 2020-04-21:
Adding it here should make it work indeed: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/startZnp.ts#L63 (in the startZnp.js) variant.

SillyDay, 2020-04-24:
@Koenkk the endpoint 1 uses ZHA(0x0504) so it cannot use another Profile ID at the same time,  is that right?

Koenkk, 2020-04-24:
Yes, I think a new endpoint has to be added.

SillyDay, 2020-04-25:
@Koenkk Thanks, but the module only send data to endpoint 1,  I contacted Ebyte, they'll release a new firmware soon. I tested the firmware and added support for E180-ZG120A/B.(a Serial port transparent transmission module)

closed 2020-04-25 by SillyDay