Koenkk/zigbee2mqtt #3393: router LED

rmalchow, 2020-04-19:
is there any way to turn off the green LED on a router CC2531? i could just user gaffer tape ... but ... seeing that it actually publishes "led_state" = true .... i thought there probably is a way. just sending led_state = false give me a converter error though ...

Fabiancrg, 2020-04-19:
If the goal is just to disable the LED, you can do it in the configuration.yaml file.

Please check the [documentation ](https://www.zigbee2mqtt.io/information/configuration.html)

rmalchow, 2020-04-20:
thank you, @Fabiancrg but this seems to be for the coordinator (also, i am using the TI dev board, and there, it doesnt even work for the coordinator) - the thing i am looking for is to switch off the LED on a CC2531 that is simply plugged into a usb wallwart and acting as a router. but it isnt that important anyways - shrink tubing or gaffer tape is fine too

rando-calrissian, 2020-04-21:
You can find some firmware here which will allow you to set the LED state as well as a bunch of other things :  https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/

closed 2020-06-28 by stale[bot]