Koenkk/zigbee2mqtt #3394: Issue with zigbee2mqtt not starting

vidarlo, 2020-04-19:
I'm using 1.12.2, and firmware 20190608 on a CC2531 device.

It was working *fine* for a week or so, but last night the stick stopped responding. A reboot solved, and I didn't worry more about it. This morning the same thing happened, and again a reboot solved it. 

It happened again this evening, but now a reboot did not resolve it.

What I've tried:
*Reboot.
*Shutdown, unplug, leave unplugged for an hour.
*Press button closest to USB port.
*Press button furthest from USB port.
*Change USB port.
*Re-plug device.

When I plug in the device (or after reboot) a red and green LED is lit. When zigbee2mqtt starts, the green led goes out. 

I've of course [captured a debug log](https://termbin.com/8yf8) as well.

vidarlo, 2020-04-20:
Reflashing firmware solved the issue.

vidarlo, 2020-04-21:
I was wrong. Reflashing firmware was a temporary fix; same error with identical error messages appeared today. Another reflash solved it, but I don't want to continue reflashing the software every day.

Koenkk, 2020-04-21:
I recommend to switch to the source routing firmware which has increased stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

vidarlo, 2020-04-22:
Great :) I'll give it a try :)

closed 2020-04-23 by vidarlo