Koenkk/zigbee2mqtt #3396: Aqara Opple turn on / off all devices accidentally

luongos, 2020-04-20:
I have 3 Aqara Opple, two of them have this problem. They worked well until some days recently. The first 2 button line (on and off). In home automations, I set up them to turn on and off a light only. But it do turn on / off all devices now.

Something I found in the log file:

`debug 2020-04-20 12:55:53: Received Zigbee message from '0x04cf8cdf3c75bd28', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-04-20 12:55:53: MQTT publish: topic 'homeassistant/device_automation/0x04cf8cdf3c75bd28/action_button_1_single/config', payload '{"automation_type":"trigger","type":"action","subtype":"button_1_single","payload":"button_1_single","topic":"zigbee2mqtt/0x04cf8cdf3c75bd28/action","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c75bd28"],"name":"0x04cf8cdf3c75bd28","sw_version":"Zigbee2mqtt 1.12.2","model":"Aqara Opple switch 2 bands (WXCJKG12LM)","manufacturer":"Xiaomi"}}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":"button_1_single"}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":""}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28/action', payload 'button_1_single'
debug 2020-04-20 12:55:53: Received Zigbee message from '0x04cf8cdf3c75bd28', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28/action', payload 'button_1_single'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":"button_1_single"}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":""}'
debug 2020-04-20 12:55:53: Received Zigbee message from '0x04cf8cdf3c75bd28', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28/action', payload 'button_1_single'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":"button_1_single"}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":""}'
debug 2020-04-20 12:55:53: Received Zigbee message from '0x04cf8cdf3c75bd28', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28/action', payload 'button_1_single'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":"button_1_single"}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":""}'
debug 2020-04-20 12:55:53: Received Zigbee message from '0x04cf8cdf3c75bd28', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28/action', payload 'button_1_single'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":"button_1_single"}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":""}'
debug 2020-04-20 12:55:53: Received Zigbee message from '0x04cf8cdf3c75bd28', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28/action', payload 'button_1_single'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":"button_1_single"}'
info  2020-04-20 12:55:53: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c75bd28', payload '{"battery":100,"voltage":3118,"linkquality":0,"action":""}'
debug 2020-04`

Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version:

luongos, 2020-04-20:
Full log file as attached. Even it doesn't work when I do repair sometimes.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4501486/log.txt)

closed 2020-04-20 by luongos

OrbitingOcelot, 2020-11-28:
@luongos Did you manage to resolve this issue? I have exaclty the same problem, switches that used to work now operate every light in the house... :-O