Koenkk/zigbee2mqtt #3398: Phillips Hue Edison Bulb 9290022413 / LWV002 - No join attempts

jbrechtel, 2020-04-20:
# Bug Report
## What happened

After resetting a Phillips Hue Edison 9290022413 bulb I do not see any join attempt in the zigbee2mqtt logs.

## What did you expect to happen

I expect to see at least a join attempt. I understand this is not a supported device but I intended to add support.

## How to reproduce it (minimal and precise)

Reset the bulb to factory settings and observe the zigbee2mqtt logs.

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version: CC2531ZNP-Prod.hex (not sure which from https://github.com/Koenkk/Z-Stack-firmware but the seller of my device gave me that name and that repo )

## Additional details

I read the FAQ on join failures and have done the following -

- `permit_join` is set to true.
- I've tried with the CC2531 very close to the bulb to prevent interference.
- I've tried TouchLink factory reset and nothing happens
- I've reset the bulb with a Hue dimmer switch (it flashes)
- I've reset the bulb through the Hue app over Bluetooth
- I've tried observing join attempts with the CC2531 on my Raspberry Pi 4 via HomeAssistant and from my desktop just running Zigbee2mqtt in docker directly. In both cases I can pair the Hue dimmer switch but see nothing from the bulb.

Notably the CC2531 does successfully pair with the Hue dimmer switch.

Regarding reflashing the firmware - I don't have the debugger needed to reflash the firmware myself but I would expect/hope to at least see a join *attempt* even if the CC2531 was at its limit. Also I've paired the remote to the CC2531 multiple times so that, at least, hints that the device hasn't hit its limit.

I actually bought another pair of bulbs as well, Hue LWA003, and all of the above is true with them too. They all work fine in the Bluetooth app and are described as having Zigbee support.

Another curious note - the "reset" from the Hue app makes the bulbs flash once but using the dimmer switch they flash 3 times.

I don't have any real reason to think this is a problem with Zigbee2mqtt per-se but I'm curious if anyone else has ran into this with newer Hue bulbs or am I just missing something?

sjorge, 2020-04-20:
Have you tried stopping z2m, removing the coordinator USB, plugging it back in and starting z2m again?

jbrechtel, 2020-04-20:
@sjorge Sorry, yes. I forgot to mention I tried that too. I rebooted the Pi as well as the procedure you just described where I just stop zigbee2mqtt, remove and re-insert the coordinator USB then start z2m.

I have the `log_level` set to `debug` and I see no messages (except about the Hue remote) after doing the factory reset procedure. I tried toggling the power of the light after factory reset as well as waiting for a long time.

It's almost like these bulbs have no Zigbee support at all?!

sjorge, 2020-04-20:
Have you tried doing the touch link reset via the coordinator? I’m on my phone so can’t pull up the link. But should be on the main page.

~ sjorge

jbrechtel, 2020-04-20:
@sjorge I have. The bulb doesn't respond to the TouchLink attempt at all. I've had the coordinator very close to the bulbs when trying. I see the TouchLink attempt in the z2m logs but it reports failure after a bit and the bulb does nothing.

sjorge, 2020-04-20:
I’m out of ideas then, hopefully someone else still has some.

jbrechtel, 2020-04-20:
Thanks for the help. I'm really confused by this.

jbrechtel, 2020-04-20:
I should also note that I have a USB extension cable moving the CC2531 about a foot away from the RP4.

I noticed that the firmware I have is older than what's in the README. I'm going to get it updated to see if that helps.

jbrechtel, 2020-04-23:
I must be missing something super basic. I ordered some Sylvania Lightify lights and they don't see to be seen by the CC2531 at all either.

I've even got an updated firmware on it now (20190608) and I see the same lack of ... well anything..from the lights.

It's so bizarre that the dimmer switch shows up but none of these lights do.

jbrechtel, 2020-04-27:
FWIW I never solved this.

I did eventually flash the sniffer firmware on my CC2531 and could see zigbee packets coming from the unseen devices during their join attempts but no idea where to go from there.

In the end the following devices would never join with the CC2531 and zigbee2mqtt

- Zen Thermostat
- Philips Hue Edison - LWV002
- Philips Hue LWA003
- Some Sylvania bulbs

The following devices did pair and work OK

- Phillips Hue dimmer switch
- Linkind motion sensor

I have since switched to a HUSBZB-1 and am using ZHA for everything.

closed 2020-04-27 by jbrechtel