Koenkk/zigbee2mqtt #3399: Pairing Gledopto GL-C-009 not working

Webworkx, 2020-04-20:
Unable to pair the gledopto GL-C-009 with zigbee2mqtt.
Pairing like described in the supported devices, lights blink also 4 times after sequence but no new devices found.

No info about new devices in the log also.
Have 2 of this devices and both the same result.

Tried pairing 20+ times, within millimeters of the CC2531 USB stick 

Use 1.12.0 version of zigbee2mqtt on rapsberry pi with domoticz.
Permit_join is set to true in configuration.yaml and domoticz zigbee2mqtt plugin

Currently 3 ikea tradfi wall outlets and 1 ikea bulb and some 2 aqara xiaomi temp sensors and 2 motion sensors on my zigbee netwerk.

CC2531 Adapter firmware = 20190608

Koenkk, 2020-04-21:
Did you check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair ?

Webworkx, 2020-04-22:
Thank you very much, i did all of that, except disconnecting the cc2531 coordinator.

After disconnecting pairing was working again.

Thanks!

closed 2020-04-23 by Webworkx