Koenkk/zigbee2mqtt #3400: Trust ZYCT-202 Remote - Failed interview - No converter available GenOff

milobt, 2020-04-20:
Trust ZYCT-202 Won't Complete Interview
Was previously paired and then only ON would work. Off command send error of no converter.
Is the modelId correct in devices.js?
My db entry doesn't mention the ZYCT-202 model number.

Database entry:
`{"id":53,"type":"EndDevice","ieeeAddr":"0x00158d0001ccab1b","nwkAddr":30579,"manufId":4151,"modelId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2080,"inClusterList":[0,4,3,6,8,4096],"outClusterList":[0,4,3,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}}},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1587400360783}`

On button:
`zigbee2mqtt:info  2020-04-20 17:49:58: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b/action', payload 'on'
zigbee2mqtt:info  2020-04-20 17:49:58: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":147,"action":"on","action_group":43808}'
zigbee2mqtt:info  2020-04-20 17:49:58: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":147,"action":""}'`

Off button:
`zigbee2mqtt:debug 2020-04-20 17:47:25: Received Zigbee message from '0x00158d0001ccab1b', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 43808
zigbee2mqtt:debug 2020-04-20 17:47:25: No converter available for 'ZYCT-202' with cluster 'genOnOff' and type 'commandOffWithEffect' and data '{"effectid":0,"effectvariant":0}'`

Up and Down:
`zigbee2mqtt:info  2020-04-20 17:49:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b/action', payload 'up-press'
zigbee2mqtt:info  2020-04-20 17:49:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":147,"action":"up-press","action_group":43808}'
zigbee2mqtt:info  2020-04-20 17:49:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":147,"action":""}'
zigbee2mqtt:info  2020-04-20 17:49:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b/action', payload 'stop'
zigbee2mqtt:info  2020-04-20 17:49:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":144,"action":"stop","action_group":43808}'
zigbee2mqtt:info  2020-04-20 17:49:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":144,"action":""}'
zigbee2mqtt:info  2020-04-20 17:49:21: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b/action', payload 'down-press'
zigbee2mqtt:info  2020-04-20 17:49:21: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":147,"action":"down-press","action_group":43808}'
zigbee2mqtt:info  2020-04-20 17:49:21: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":147,"action":""}'
zigbee2mqtt:info  2020-04-20 17:49:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b/action', payload 'stop'
zigbee2mqtt:info  2020-04-20 17:49:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":141,"action":"stop","action_group":43808}'
zigbee2mqtt:info  2020-04-20 17:49:22: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":141,"action":""}'`

## Debug Info
zigbee2mqtt version: 1.12.2
CC26X2R1 firmware version: 20191106

milobt, 2020-04-22:
db entry:

`{"id":53,"type":"EndDevice","ieeeAddr":"0x00158d0001ccab1b","nwkAddr":43471,"manufId":4151,"modelId":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,2],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1587547197822}`

---

Logs:

```
info  2020-04-22 10:16:09: MQTT publish: topic 'homeassistant/sensor/0x00158d0001ccab1b/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x00158d0001ccab1b","json_attributes_topic":"zigbee2mqtt/0x00158d0001ccab1b","name":"0x00158d0001ccab1b_action","unique_id":"0x00158d0001ccab1b_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0001ccab1b"],"name":"0x00158d0001ccab1b","sw_version":"Zigbee2mqtt 1.12.2-dev","model":"Remote control (ZYCT-202)","manufacturer":"Trust"},"availability_topic":"zigbee2mqtt/0x00158d0001ccab1b/availability"}'
info  2020-04-22 10:16:09: MQTT publish: topic 'homeassistant/sensor/0x00158d0001ccab1b/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d0001ccab1b","json_attributes_topic":"zigbee2mqtt/0x00158d0001ccab1b","name":"0x00158d0001ccab1b_linkquality","unique_id":"0x00158d0001ccab1b_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0001ccab1b"],"name":"0x00158d0001ccab1b","sw_version":"Zigbee2mqtt 1.12.2-dev","model":"Remote control (ZYCT-202)","manufacturer":"Trust"},"availability_topic":"zigbee2mqtt/0x00158d0001ccab1b/availability"}'
error 2020-04-22 10:16:18: Failed to configure '0x00158d0001ccab1b', attempt 1 (Error: Bind 0x00158d0001ccab1b/1 genOnOff from '0x00124b001ca1634e/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
debug 2020-04-22 10:19:20: Received Zigbee message from '0x00158d0001ccab1b', type 'commandOffWithEffect', cluster 'genOnOff', data '{"effectid":0,"effectvariant":0}' from endpoint 1 with groupID 43808
info  2020-04-22 10:19:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":150,"action":"off","action_group":43808}'
info  2020-04-22 10:19:20: Configuring '0x00158d0001ccab1b'
info  2020-04-22 10:19:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b', payload '{"linkquality":150,"action":""}'
info  2020-04-22 10:19:20: MQTT publish: topic 'homeassistant/device_automation/0x00158d0001ccab1b/action_off/config', payload '{"automation_type":"trigger","type":"action","subtype":"off","payload":"off","topic":"zigbee2mqtt/0x00158d0001ccab1b/action","device":{"identifiers":["zigbee2mqtt_0x00158d0001ccab1b"],"name":"0x00158d0001ccab1b","sw_version":"Zigbee2mqtt 1.12.2-dev","model":"Remote control (ZYCT-202)","manufacturer":"Trust"}}'
info  2020-04-22 10:19:20: MQTT publish: topic 'zigbee2mqtt/0x00158d0001ccab1b/action', payload 'off'
error 2020-04-22 10:19:30: Failed to configure '0x00158d0001ccab1b', attempt 2 (Error: Bind 0x00158d0001ccab1b/1 genOnOff from '0x00124b001ca1634e/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Koenkk, 2020-04-22:
Seems to the converter warning is fixed now.

Regarding configure, you can try to trigger it manually via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure . Make sure to wake the device up right before doing this, otherwise it will fail.

closed 2020-04-24 by milobt