Koenkk/zigbee2mqtt #3401: 2 x IKEA E1524/E1810 remotes, one has no battery status

Anjerlaan, 2020-04-20:
# Bug Report
## What happened
I have 2 IKEA E1524/E1810 remotes.
One is showing battery status and one is not

```
info  2020-04-20 22:28:39: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":36,"update_available":false}'
info  2020-04-20 22:28:39: MQTT publish: topic 'zigbee2mqtt/Remote_Tuinlicht', payload '{"update_available":false,"linkquality":36,"battery":74}'
```

Twice I performed a "zigbee2mqtt/bridge/configure", where the remote was configured succesfully according to the log:

`info  2020-04-20 22:41:30: Successfully configured 'Remote_E1524-E1810'
debug 2020-04-20 22:42:01: Received Zigbee message from 'Keuken_Router', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 8 with groupID 0
info  2020-04-20 22:42:01: MQTT publish: topic 'zigbee2mqtt/Keuken_Router', payload '{"state":true,"led_state":true,"linkquality":44}'
debug 2020-04-20 22:42:26: Received Zigbee message from 'Remote_E1524-E1810', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 100
info  2020-04-20 22:42:26: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":31,"update_available":false,"action":"toggle"}'`

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version: 20190608

ChrisSantiago82, 2020-04-20:
I have the same issus with a couple of Ikea sensors.
I'm assuming it's coming from different firmware versions. Maybe it's the same with the remotest?

Koenkk, 2020-04-21:
It can take up to 24 hours before the remote is sending it's battery status.

Anjerlaan, 2020-04-21:
Okay...24 hours later...still no battery level included.
In my logs I see that in 1.12.0 in March this already was the case.
I can’t remember if this ever was an issue.
The reason I noticed was during creation of node-red battery level notification flows

Koenkk, 2020-04-22:
Can you try to reconfigure the device? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure Make sure to wake up the device right before doing that.

I would expect the battery percentage after 24 hours.

Anjerlaan, 2020-04-22:
> 
> 
> Can you try to reconfigure the device? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure Make sure to wake up the device right before doing that.
> 
> I would expect the battery percentage after 24 hours.

Okay re-configured it again:

`info  2020-04-22 18:59:11: Successfully configured 'Remote_E1524-E1810'
debug 2020-04-22 18:59:29: Saving state to file /opt/zigbee2mqtt/data/state.json`

I already configured it 2 times, 36 hours ago....the log reported both times: configured succesfully
Then you may expect that I woke up the device correcty, or not?

ChrisSantiago82, 2020-04-22:
Can you check if there is an firmware update available for that device? (OTA).
I have the same problem with my ikea devices if there is an Update available it won't show the battery status. The once that are uptodate, will show it.

Anjerlaan, 2020-04-22:
They both have the last firmware

Anjerlaan, 2020-04-22:
I have a CC2531 as Coördinator and a CC2530 as a router.
I may have configured the Remote (with no battery level) with the router.
Could that be an issue?

mr2c12, 2020-04-23:
In my setup, I've always experienced some incoherence regarding battery reporting. Sometimes it takes a long time to get a battery level. But, at some point, you get that.

I have a similar setup, whereas the IKEA remote was connected to a CC2530 router.

Anjerlaan, 2020-04-23:
Well according to my logs, there was already no battery level 6 weeks ago. The other identical remote, which i have since 3 weeks, has no problem from the start I configured it.

Anjerlaan, 2020-04-25:
48hours later, still no battery level

Koenkk, 2020-04-26:
Can you try changing the entry in `devices.js` to 

```js
{
    zigbeeModel: ['TRADFRI remote control'],
    model: 'E1524/E1810',
    description: 'TRADFRI remote control',
    supports:
        'toggle, arrow left/right click/hold/release, brightness up/down click/hold/release',
    vendor: 'IKEA',
    fromZigbee: [
        fz.cmdToggle, fz.E1524_arrow_click, fz.E1524_arrow_hold, fz.E1524_arrow_release,
        fz.E1524_brightness_up_click, fz.E1524_brightness_down_click, fz.E1524_brightness_up_hold,
        fz.E1524_brightness_up_release, fz.E1524_brightness_down_hold, fz.E1524_brightness_down_release,
        fz.battery_not_divided, fz.E1524_hold,
    ],
    toZigbee: [],
    ota: ota.tradfri,
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        // See explanation in E1743, only applies to E1810 (for E1524 it has no effect)
        // https://github.com/Koenkk/zigbee2mqtt/issues/2772#issuecomment-577389281
        await endpoint.bind('genOnOff', defaultBindGroup);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await configureReporting.batteryPercentageRemaining(endpoint);
        await configureReporting.batteryVoltage(endpoint);
    },
},
```

Make sure it configures correctly after this change. See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to get to `devices.js`.

Anjerlaan, 2020-04-26:
> Can you try changing the entry in `devices.js` to
>
> ```js
> {
>     zigbeeModel: ['TRADFRI remote control'],
>     model: 'E1524/E1810',
>     description: 'TRADFRI remote control',
>     supports: [...]

thanx, I’ll try when I have the time!
Do I need to set “permit_join: true”, before modifying devices.js?

Koenkk, 2020-04-26:
No

Anjerlaan, 2020-04-26:
@Koenkk 

1. Modified devices.js
2. Restarted Zigbee2mqtt
3. Forgot to wake up the 2 remote devices ;-)
4. Configuration Error
5. Restarted Zigbee2mqtt while waking up devices
6. Configuration successful for both devices!
7. Remote_Tuinlicht has battery level (which already was the case)....and Remote_E1524-E1810 has no battery level.

next step I did a "zigbee2mqtt/bridge/configure" only for Remote_E1524-E1810...still no battery level (although I should wait 24 hours to be sure)

Basically: Nothing changed

Here is the log:
https://pastebin.com/hyypVbCF

What is the function of this line you added?:

`await configureReporting.batteryVoltage(endpoint);`

Koenkk, 2020-04-28:
Can you try replacing your `devices.js` with the following: https://github.com/Koenkk/zigbee-herdsman-converters/blob/E1524_battery/devices.js

After a successful configure it should report its battery max every 5 seconds (for testing purposes). Does it do that?

Anjerlaan, 2020-04-28:
> 
> 
> Can you try replacing your `devices.js` with the following: https://github.com/Koenkk/zigbee-herdsman-converters/blob/E1524_battery/devices.js
> 
> After a successful configure it should report its battery max every 5 seconds (for testing purposes). Does it do that?

@Koenkk 

I have replaced devices.js with the file in your link but then zigbee2mqtt doesn't start.
With the previous devices.js file, zigbee2mqtt starts normally. (I put the previous file back)

When I watched the Log of the "previous" devices.js I noticed the following lines:

```
debug 2020-04-28 19:40:14: Received Zigbee message from 'Remote_E1524-E1810', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 100
info  2020-04-28 19:40:14: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":36,"update_available":false,"action":"toggle"}'
debug 2020-04-28 19:40:15: Received Zigbee message from 'Remote_Tuinlicht', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 200
info  2020-04-28 19:40:15: MQTT publish: topic 'zigbee2mqtt/Remote_Tuinlicht', payload '{"update_available":false,"linkquality":42,"battery":21,"voltage":2600,"action":"toggle"}'
```
 The Voltage line, you added in the previous reply is showing a voltage value for "Remote_Tuinlicht"
Of course the "problem remote" does nothing.

`await configureReporting.batteryVoltage(endpoint);`

Anjerlaan, 2020-04-28:
Okay...I removed the battery temporary (put out/put in) from the "Remote_E1524-E1810" at 20:10h
The LOG showed that the Remote announced itself, but no Battery and Voltage status:

```
debug 2020-04-28 20:11:19: Device 'Remote_E1524-E1810' announced itself
info  2020-04-28 20:11:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"Remote_E1524-E1810"}}'
debug 2020-04-28 20:11:19: Device 'Remote_E1524-E1810' announced itself
info  2020-04-28 20:11:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"Remote_E1524-E1810"}}'
debug 2020-04-28 20:11:23: Received Zigbee message from 'Remote_E1524-E1810', type 'commandStepWithOnOff', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":43,"transtime":5}' from endpoint 1 with groupID 100
info  2020-04-28 20:11:23: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":44,"update_available":false,"action":"brightness_up_click"}'
debug 2020-04-28 20:11:28: Received Zigbee message from 'Remote_E1524-E1810', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 100
info  2020-04-28 20:11:28: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":36,"update_available":false,"action":"toggle"}'
```

And then 1 hour later suddenly????:
```

debug 2020-04-28 21:12:53: Received Zigbee message from 'Remote_E1524-E1810', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":60,"batteryVoltage":28}' from endpoint 1 with groupID 0
info  2020-04-28 21:12:53: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":39,"update_available":false,"battery":60,"voltage":2800}'
```
Could it be that removing the battery was the trigger?

Koenkk, 2020-04-29:
Cool, yes we also see that for OTA, I've added this to the docs.

Can we close this issue?

Anjerlaan, 2020-04-29:
@Koenkk 
Yes you can close it!
Summary:

This test "devices.js" is not starting zigbee2mqtt at all
> 
> 
> Can you try replacing your `devices.js` with the following: https://github.com/Koenkk/zigbee-herdsman-converters/blob/E1524_battery/devices.js
> 
> After a successful configure it should report its battery max every 5 seconds (for testing purposes). Does it do that?

Changing the "devices.js" file of build 1.12.2 with these lines is adding "Voltage" status, which is nice because the domoticz-zigbee2mqtt plugin add's for each IKEA Remote E1524/E1810 also a Voltage device, which never worked until you added this to devices.js:

> Can you try changing the entry in `devices.js` to
>
> ```js
> {
>     zigbeeModel: ['TRADFRI remote control'],
>     model: 'E1524/E1810',
>     description: 'TRADFRI remote control',
>     supports: [...]
Thanx for the Support!

Koenkk, 2020-04-29:
Thanks, the preferred way is battery percentage over voltage so let's leave it as is.

closed 2020-04-29 by Koenkk