Koenkk/zigbee2mqtt #3402: Strange behavior of Eurotronic Spirit thermostats: erratic temp readings

CodeFinder2, 2020-04-20:
Hi,

this is not really a bug report; I rather observed strange behavior with my Eurotronic Spirit zigbee thermostats, log:
```
zigbee2mqtt:debug 2020-04-21 00:38:39: Device 'kitchen/thermostat' announced itself
zigbee2mqtt:info  2020-04-21 00:38:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"kitchen/thermostat"}}'
zigbee2mqtt:debug 2020-04-21 00:38:40: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"16387":2000}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:40: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":17,"occupied_heating_setpoint":15,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":75,"last_seen":"2020-04-20T22:38:40.904Z","current_heating_setpoint":20,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
zigbee2mqtt:debug 2020-04-21 00:38:40: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"unoccupiedHeatingSetpoint":2000,"occupiedHeatingSetpoint":2000,"pIHeatingDemand":255,"localTemp":2000}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:40: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":20,"occupied_heating_setpoint":20,"unoccupied_heating_setpoint":20,"pi_heating_demand":100,"linkquality":75,"last_seen":"2020-04-20T22:38:40.914Z","current_heating_setpoint":20,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
zigbee2mqtt:debug 2020-04-21 00:38:42: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":1700,"pIHeatingDemand":0,"occupiedHeatingSetpoint":1500,"unoccupiedHeatingSetpoint":1600}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:42: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":17,"occupied_heating_setpoint":15,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":75,"last_seen":"2020-04-20T22:38:42.738Z","current_heating_setpoint":20,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
zigbee2mqtt:debug 2020-04-21 00:38:42: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"16387":1500,"16392":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:42: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":17,"occupied_heating_setpoint":15,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":75,"last_seen":"2020-04-20T22:38:42.805Z","current_heating_setpoint":15,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
zigbee2mqtt:debug 2020-04-21 00:38:42: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"unoccupiedHeatingSetpoint":1600,"occupiedHeatingSetpoint":1500,"pIHeatingDemand":0,"localTemp":1700}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:42: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":17,"occupied_heating_setpoint":15,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":75,"last_seen":"2020-04-20T22:38:42.817Z","current_heating_setpoint":15,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
zigbee2mqtt:debug 2020-04-21 00:38:43: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":1700,"pIHeatingDemand":0,"occupiedHeatingSetpoint":1500,"unoccupiedHeatingSetpoint":1600}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:43: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":17,"occupied_heating_setpoint":15,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":75,"last_seen":"2020-04-20T22:38:43.059Z","current_heating_setpoint":15,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
zigbee2mqtt:debug 2020-04-21 00:38:44: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":1700,"pIHeatingDemand":0,"occupiedHeatingSetpoint":1500,"unoccupiedHeatingSetpoint":1600}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:44: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":17,"occupied_heating_setpoint":15,"unoccupied_heating_setpoint":16,"pi_heating_demand":0,"linkquality":75,"last_seen":"2020-04-20T22:38:44.617Z","current_heating_setpoint":15,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
```

If this is normal, this can simply be closed but since I haven't seen this before, I thought its worth reporting it. Happens from time to time and it also affects other devices. Unfortunately, I don't know how it can be triggered / reproduced.

I am talking about these lines in the log:
```
zigbee2mqtt:debug 2020-04-21 00:38:40: Received Zigbee message from 'kitchen/thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"unoccupiedHeatingSetpoint":2000,"occupiedHeatingSetpoint":2000,"pIHeatingDemand":255,"localTemp":2000}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-04-21 00:38:40: MQTT publish: topic 'zigbee2mqtt/kitchen/thermostat', payload '{"local_temperature":20,"occupied_heating_setpoint":20,"unoccupied_heating_setpoint":20,"pi_heating_demand":100,"linkquality":75,"last_seen":"2020-04-20T22:38:40.914Z","current_heating_setpoint":20,"eurotronic_host_flags":{"mirror_display":false,"boost":false,"window_open":false,"child_protection":false},"system_mode":"auto","eurotronic_system_mode":1,"eurotronic_error_status":0,"battery":40,"battery_low":0}'
```
causing an erratic change in Home Assistant's web UI. Again, not really an issue but seems also incorrect. Or is it a bug on the device side?! I have not changed anything via the device's buttons. Also, no automations were active for the device and no commands have been sent manually (as you can all see in the log).

Version of Zigbee2Mqtt: 1.12.2 (via HA v0.108.6
Coordinator version: 20200328
HW: CC1352P-2

swissbyte, 2020-04-22:
For further diagnostics, you could enable herdsman debug mode and post its logfiles. This will help to figer out where these messages come from and why.

CodeFinder2, 2020-04-22:
Debug mode is already enabled? 🙈🤔

CodeFinder2, 2020-05-06:
@Koenkk any ideas or should I simply close this? 🙈

Koenkk, 2020-05-07:
@CodeFinder2 looks it's something on the device side, it reports all it's state after an announce on the network.

To work around it you could use `debounce: 4`, but this comes with the downside that all messages from the device takes 4 seconds longer to arrive. https://www.zigbee2mqtt.io/information/configuration.html#all-devices

CodeFinder2, 2020-05-07:
Okay, thanks! Will leave it as it is since its rather a cosmetic issue. ☺️

closed 2020-05-07 by CodeFinder2