Koenkk/zigbee2mqtt #3403: Strange warnings for 'Aqara double key wired wall switch (QBKG12LM)'

quarcko, 2020-04-21:
Some time ago i noticed strange warnings (don't remember from which update it started)
Now i am on latest Z2M and also latest HA 0.108.6

The devices in question are QBKG12LM (Double switches)
I also in home have Single switches and Mijia smart outlets, which are not affected:

Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/0x00158d000249790c/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d000249790c'], 'name': '0x00158d000249790c', 'sw_version': 'Zigbee2mqtt 1.12.2', 'model': 'Aqara double key wired wall switch (QBKG12LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover hass, config, config_entry, discovery_data File "/usr/src/homeassistant/homeassistant/components/mqtt/device_trigger.py", line 174, in async_setup_trigger config = TRIGGER_DISCOVERY_SCHEMA(config) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']

This is a warning inside HA logs (not Z2M) but they are coming from Z2M discovery payloads?
Also strange thing noticed:
All these devices (Double wall switch, single wall switch, smart outlet) all of them have temperature
monitoring sensor, right?
But ONLY QBKG12LM actually have a separate sensor in MQTT discovery payloads as 'entity'.
All the rest of them do not have it in discovery payloads?

So it seems double switches are treated a bit differently by Z2M than the rest?
although containing same information?

Koenkk, 2020-04-21:
This is a MQTT device trigger which only button devices have (https://www.home-assistant.io/integrations/device_trigger.mqtt/). 

I expect it to have the subtype (https://github.com/Koenkk/zigbee2mqtt/blob/864e8b26d5cb6ef951299aea21423ffb25304b1d/lib/extension/homeassistant.js#L1607). Could you post the Zigbee2mqtt log at the moment this error happens?

acelauwc, 2020-04-23:
I have the same problem with this switch as well. strangely only this specific switch, other switches of the same model is ok.  Lastest Z2M and HA 0.108 as well.

Log Details (ERROR)
Logger: homeassistant.util.logging
First occured: 9:40:40 AM (1 occurences)
Last logged: 9:40:40 AM

Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/ws_commontoilet/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d00022ad621'], 'name': 'ws_commontoilet', 'sw_version': 'Zigbee2mqtt 1.12.0', 'model': 'Aqara double key wired wall switch (QBKG12LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/mqtt/device_automation.py", line 34, in async_discover
    await device_trigger.async_setup_trigger(
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/mqtt/device_trigger.py", line 168, in async_setup_trigger
    config = TRIGGER_DISCOVERY_SCHEMA(config)
  File "/srv/homeassistant/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/srv/homeassistant/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/srv/homeassistant/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']

_**zb2mqtt log for the specific switch:**_
info  2020-04-23 09:39:22: MQTT publish: topic 'zigbee2mqtt/ws_commontoilet', payload '{"power":35.07,"consumption":54.04,"temperature":34,"linkquality":55,"state_left":"ON","state_right":"ON","operation_mode_left":"decoupled","operation_mode_right":"decoupled"}'

**_zb2mqtt log around the time this occur:_**
info  2020-04-23 09:39:56: MQTT publish: topic 'zigbee2mqtt/ws_masterbedroom', payload '{"power":1.67,"consumption":59.27,"temperature":32,"linkquality":123,"state_left":"ON","state_right":"OFF","operation_mode_left":"decoupled"}'
info  2020-04-23 09:40:01: MQTT publish: topic 'zigbee2mqtt/ms_entrance', payload '{"battery":100,"voltage":3025,"illuminance":0,"linkquality":120,"occupancy":false,"illuminance_lux":0}'
info  2020-04-23 09:40:38: MQTT publish: topic 'zigbee2mqtt/0x00124b00194a0159', payload '{"state":true,"led_state":false,"linkquality":120}'
info  2020-04-23 09:40:44: MQTT publish: topic 'zigbee2mqtt/ws_mastertoilet', payload '{"power":2.4,"linkquality":123,"consumption":83.42,"temperature":31,"state_left":"ON","state_right":"ON"}'
info  2020-04-23 09:41:01: MQTT publish: topic 'zigbee2mqtt/ws_mastertoilet', payload '{"power":9.62,"linkquality":123,"consumption":83.42,"temperature":31,"state_left":"ON","state_right":"ON"}'
info  2020-04-23 09:41:13: MQTT publish: topic 'zigbee2mqtt/sw_masterbedroom_cony', payload '{"linkquality":123,"battery":100,"voltage":3035}'
info  2020-04-23 09:41:38: MQTT publish: topic 'zigbee2mqtt/0x00124b00194a0159', payload '{"state":true,"led_state":false,"linkquality":63}'
info  2020-04-23 09:42:23: MQTT publish: topic 'homeassistant/device_automation/0x00158d0001e06e7a/click_right/config', payload '{"automation_type":"trigger","type":"click","subtype":"right","payload":"right","topic":"zigbee2mqtt/sw_kitchen_light/click","device":{"identifiers":["zigbee2mqtt_0x00158d0001e06e7a"],"name":"sw_kitchen_light","sw_version":"Zigbee2mqtt 1.12.2","model":"Aqara double key wireless wall switch (WXKG02LM)","manufacturer":"Xiaomi"}}'

quarcko, 2020-04-27:
I noticed in MQTT Explorer, that on discovery topics "homeassistant/device_automation"
only these double switches have "click_undefined":
{"automation_type":"trigger","type":"click","topic":"zigbee2mqtt/0x00158d0001bf9c83/click","device":{"identifiers":["zigbee2mqtt_0x00158d0001bf9c83"],"name":"0x00158d0001bf9c83","sw_version":"Zigbee2mqtt 1.12.2","model":"Aqara double key wired wall switch (QBKG12LM)","manufacturer":"Xiaomi"}}

which lack "subtype". As these messages are "retained" most probably they are still here
from older version of zigbee2mqtt software.

I will try now deleting these messages and restart HA.

For me these messages always appear on HA restart, so most probably not z2m is sending them
but HA is taking them from retained messages from mosquitto.

quarcko, 2020-04-27:
No i was wrong, after restarting both Z2M and HA, these "click_undefined" returned back,
and there are again those warnings...

What is this "click_undefined" ?

AlexeyHimki, 2020-05-12:
Have same error after each reboot. Z2M 1.13.0, HA 0.109.2

Logger: homeassistant.util.logging
Source: util/logging.py:162
First occurred: 22:25:21 (1 occurrences)
Last logged: 22:25:21

Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/tualet.wired_switch/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0004089cad'], 'name': 'tualet.wired_switch', 'sw_version': 'Zigbee2mqtt 1.12.2', 'model': 'Aqara double key wired wall switch (QBKG12LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/home/pi/homeassistant/lib/python3.7/site-packages/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover hass, config, config_entry, discovery_data File "/home/pi/homeassistant/lib/python3.7/site-packages/homeassistant/components/mqtt/device_trigger.py", line 174, in async_setup_trigger config = TRIGGER_DISCOVERY_SCHEMA(config) File "/home/pi/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/home/pi/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/home/pi/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']

Koenkk, 2020-05-14:
@quarcko could you publish the zigbee2mqtt debug log at the moment `click_undefined` is published? 

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

AlexeyHimki, 2020-05-15:
My log is here. There is strange version in HA log - Zigbee2mqtt 1.12.2. But my Z2M version is 1.13.0

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4633459/log.txt)

Logger: homeassistant.util.logging
Source: util/logging.py:162
First occurred: 12:12:15 (1 occurrences)
Last logged: 12:12:15

Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/tualet.wired_switch/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0004089cad'], 'name': 'tualet.wired_switch', 'sw_version': 'Zigbee2mqtt 1.12.2', 'model': 'Aqara double key wired wall switch (QBKG12LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/home/pi/homeassistant/lib/python3.7/site-packages/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover hass, config, config_entry, discovery_data File "/home/pi/homeassistant/lib/python3.7/site-packages/homeassistant/components/mqtt/device_trigger.py", line 174, in async_setup_trigger config = TRIGGER_DISCOVERY_SCHEMA(config) File "/home/pi/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/home/pi/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/home/pi/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']

Koenkk, 2020-05-15:
I don't see the relevant part where zigbee2mqtt publishes the information causing the error in HA. 

I think you can trigger it by starting zigbee2mqtt fresh and pressing a button on the switch. Please provide the debug logging of this.

iStitch07, 2020-05-21:
Hello. Have this issue with Aqara Relay, what is strange
```
Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/Aqara Relay/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0003e82ee5'], 'name': 'Aqara Relay', 'sw_version': 'Zigbee2mqtt 1.13.0-dev', 'model': 'Aqara wireless relay controller (LLKZMK11LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover hass, config, config_entry, discovery_data File "/usr/src/homeassistant/homeassistant/components/mqtt/device_trigger.py", line 174, in async_setup_trigger config = TRIGGER_DISCOVERY_SCHEMA(config) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']
```

MoonlightGuest, 2020-06-12:
Hello,
I also have such error in the log with Aqara relays that should not have any "click" trigger
`2020-06-12 13:47:19 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/Relay Garage/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d000444d997'], 'name': 'Relay Garage', 'sw_version': 'Zigbee2mqtt 1.13.1', 'model': 'Aqara wireless relay controller (LLKZMK11LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover
    hass, config, config_entry, discovery_data
  File "/usr/src/homeassistant/homeassistant/components/mqtt/device_trigger.py", line 178, in async_setup_trigger
    config = TRIGGER_DISCOVERY_SCHEMA(config)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']

2020-06-12 13:47:19 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/Relay 1st Floor/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d000445413c'], 'name': 'Relay 1st Floor', 'sw_version': 'Zigbee2mqtt 1.13.1', 'model': 'Aqara wireless relay controller (LLKZMK11LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover
    hass, config, config_entry, discovery_data
  File "/usr/src/homeassistant/homeassistant/components/mqtt/device_trigger.py", line 178, in async_setup_trigger
    config = TRIGGER_DISCOVERY_SCHEMA(config)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']`

There are related zigbee2mqtt log entries. But there are were no relay clicks at that time.
`zigbee2mqtt:info  2020-06-12 14:10:21: MQTT publish: topic 'zigbee2mqtt/Relay Garage', payload '{"power":0.1,"consumption":0.13,"temperature":40,"linkquality":52,"state_l1":"OFF","state_l2":"ON","click":""}'
zigbee2mqtt:info  2020-06-12 14:10:21: MQTT publish: topic 'zigbee2mqtt/Relay Garage/click', payload 'undefined'
`

ALaDoffe, 2020-06-24:
Hello,

Same error for me with LLKZMK11LM at HA startup.
But no issue with the device.

Exception in async_discover when dispatching 'mqtt_discovery_new_device_automation_mqtt': ({'automation_type': 'trigger', 'type': 'click', 'topic': 'zigbee2mqtt/0x00158d00042ac001/click', 'device': {'identifiers': ['zigbee2mqtt_0x00158d00042ac001'], 'name': '0x00158d00042ac001', 'sw_version': 'Zigbee2mqtt 1.13.1-dev', 'model': 'Aqara wireless relay controller (LLKZMK11LM)', 'manufacturer': 'Xiaomi'}, 'platform': 'mqtt'},) Traceback (most recent call last): File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/mqtt/device_automation.py", line 42, in async_discover hass, config, config_entry, discovery_data File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/mqtt/device_trigger.py", line 178, in async_setup_trigger config = TRIGGER_DISCOVERY_SCHEMA(config) File "/srv/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/srv/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/srv/homeassistant/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: required key not provided @ data['subtype']

Koenkk, 2020-06-24:
@ALaDoffe there are some fix in Zigbee2mqtt 1.14.0 so please first try this.

ALaDoffe, 2020-06-25:
Hi Koenkk,

I was already in a high dev version of Z2M and latest HA 0.111.4
info  2020-06-22 17:53:14: Starting zigbee2mqtt version 1.14.0-dev (commit #76745ca)

I have just updated Z2M to Latest-dev and still the same.
info  2020-06-25 09:03:36: Starting zigbee2mqtt version 1.14.0-dev (commit #27972fb)

For information, my config below:
RPI 3b+
HA and Z2M in Venv.
CC2531.

Regards

Koenkk, 2020-06-27:
The log shows `'sw_version': 'Zigbee2mqtt 1.13.1-dev'` so it is probably and old retained message. Try to clear this retained MQTT message by publishing to `zigbee2mqtt/0x00158d00042ac001/click` and empty payload with retain enable (you can use e.g. MQTT.fx) for this.

ALaDoffe, 2020-07-26:
Hi Koenkk,

Sorry for the delay.
Indeed it was. Deleting retain message resolved the issue.
Thank you very much for your response.

Regards

closed 2020-07-26 by Koenkk