Koenkk/zigbee2mqtt #3404: Support for Philips Hue Centura (5045148P7)

yorickri, 2020-04-21:
I'm using a set of Philips Hue Centura spots (https://www2.meethue.com/nl-nl/p/hue-white-and-color-ambiance-centura-inbouwspot/5055148P7) but unfortunately they don't seem to be supported as of yet.

I've tried to follow the "How to support new devices" page, without any luck. As I can't control the device (unsupported), i can't trigger any action on it. When i manually try to send something using MQTTExplorer, the "unsupported" message is the only thing shown.

**Output from log txt when initially pairing one of the spots**
info  2020-04-21 09:03:30: Device '0x00178801084a47de' joined
info  2020-04-21 09:03:30: Starting interview of '0x00178801084a47de'
info  2020-04-21 09:03:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00178801084a47de"}}'
info  2020-04-21 09:03:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00178801084a47de"}}'
debug 2020-04-21 09:03:31: Device '0x00178801084a47de' announced itself
info  2020-04-21 09:03:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00178801084a47de"}}'
debug 2020-04-21 09:03:50: Device '0x00178801084a47de' announced itself
info  2020-04-21 09:03:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00178801084a47de"}}'
debug 2020-04-21 09:04:07: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"modelId":"5045148P7"}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:07: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:07: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:07: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Philips"}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:07: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:07: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:07: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Philips"}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:07: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:07: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:07: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:07: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:07: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200124"}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-04-21 09:04:08: Received Zigbee message from '0x00178801084a47de', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1.65.9_hB3217DF4"}' from endpoint 11 with groupID 0
warn  2020-04-21 09:04:08: Received message from unsupported device with Zigbee model '5045148P7'
warn  2020-04-21 09:04:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2020-04-21 09:04:09: Successfully interviewed '0x00178801084a47de', device has successfully been paired
warn  2020-04-21 09:04:09: Device '0x00178801084a47de' with Zigbee model '5045148P7' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2020-04-21 09:04:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00178801084a47de","supported":false}}'

**Row from database.db for one of the spots**
{"id":14,"type":"Router","ieeeAddr":"0x00178801084a47de","nwkAddr":64144,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"5045148P7","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64514,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"5045148P7","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20200124","swBuildId":"1.65.9_hB3217DF4"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20200124","swBuildId":"1.65.9_hB3217DF4","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1587452648913}

sjorge, 2020-04-21:
You can try adding this to devices.js

```
    {
        zigbeeModel: ['5045148P7'],
        model: '5045148P7',
        vendor: 'Philips',
        description: 'Hue Centura',
        extend: hue.light_onoff_brightness_colortemp_colorxy,
        ota: ota.zigbeeOTA,
    },
```

yorickri, 2020-04-21:
That seemed to have to have done the trick. All working like a charm by adding that to devices.js!

sjorge, 2020-04-21:
I've created a pull request against zigbee-herdsman-converters, once merged it should be in the next release.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/1171

yorickri, 2020-04-21:
Thanks for that @sjorge . Would i still have to manually adjust anything in order have the device auto-discovered in Home Assistant? Or would this be taken care of when the changes have been merged and my addon in Home Assistant gets updated?

sjorge, 2020-04-21:
That would probably indeed still require a change to homeassistant.js in this repo.

closed 2020-04-23 by yorickri

Dylantje, 2020-08-04:
Is this device hardcore supported?
Or how do i get this device working, without this part?

[ do not now how to update the z2m to domoticz plugin.. ]