Koenkk/zigbee2mqtt #3405: Using z2m to reset CC2531 crashes z2m

Westcott1, 2020-04-21:
This happened with the latest dev version -

**zigbee2mqtt:info  2020-04-21 10:46:45: Soft resetted ZNP**

(node:23418) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":168,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,21,2,11,254,0]}}' failed with status '2' (expected '0')
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
(node:23418) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:23418) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Koenkk, 2020-04-21:
The soft reset feature indeed doesn't work correctly, it's a legacy thing which should be removed. Why did you use it?

Westcott1, 2020-04-21:
To save having to stop z2m, unplug the stick, re-plug it, and restart z2m.
It seemed like a good idea at the time!

Koenkk, 2020-04-21:
It does indeed, but why do you need to re-plug? (I prefer to solve the root problem).

Westcott1, 2020-04-21:
I agree, it's best to solve the actual problem.
However, this seems to be the only way to get some devices to pair.

Koenkk, 2020-04-21:
Is it the same as #3177 ? Maybe try the firmware from: https://github.com/Koenkk/zigbee2mqtt/issues/3177#issuecomment-612094951

Westcott1, 2020-04-21:
Exactly that!
I'll flash this new version, taking note of the comments.
Thanks for everything z2m-wise.

closed 2020-04-21 by Westcott1