Koenkk/zigbee2mqtt #3407: Issue with pairing IKEA repeater E1746 with CC2531

MacieIKS, 2020-04-21:
# Bug Report
## What happened
Hello, I am trying to include IKEA Tradfri Repeater (E1746). My setup is CC2531 with firmware zStack12 - 20190608. I was hitting device limitation (20 devices working perfectly fine) so I unpaired few ZigBee nodes to leave the space. When I am trying to pair this repeater, according to the logs I see it is being successfully paired:
```
info  2020-04-21 19:12:08: Successfully interviewed '0x000d6ffffe9a8752', device has successfully been paired
```

however after next step: 
```
info  2020-04-21 19:12:08: Device '0x000d6ffffe9a8752' is supported, identified as: IKEA TRADFRI signal repeater (E1746)
info  2020-04-21 19:12:08: Configuring '0x000d6ffffe9a8752'
```

I see it is failing with the following:
```
error 2020-04-21 19:12:18: Failed to configure '0x000d6ffffe9a8752', attempt 1 (Error: Bind 0x000d6ffffe9a8752/1 genBasic from '0x00124b0018e210d6/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

```
I was trying to do this on latest edge and latest stable, also started with default network settings (including reinstallation in fresh directory and moving to different network key) but results are the same. Perhaps someone faced similar like me issues and got some workaround?

## How to reproduce it (minimal and precise)
Fresh installation Z2M stable / edge, default network settings.

## Debug Info
Zigbee2mqtt version: latest stable / latest edge
Adapter hardware: CC2531
Adapter firmware version: zStack12 - 20190608

Logs with zigbee_herdsman_debug: true:
[https://pastebin.com/JzJwbetN](url)

Thank you for your inputs,
Cheers

Koenkk, 2020-04-21:
Can you check if there is an update available for your repeater? https://www.zigbee2mqtt.io/information/ota_updates.html

MacieIKS, 2020-04-22:
For some reason, when this exception is being triggered it hangs whole network and only way to restore it is to restart process with CC2531 stick reinsertion. 
Anyway I setup another network (previously deleting z2m folder and plugin reinstallation) on channel 25 and performed OTA just after moment when device was presented in front of the controller. As you can see from the logs I was trying to make it twice (07:46:57 and 07:47:07), something showed up after 2nd time, but after few seconds this exception triggered and nothing more I could do. Of course I was trying to perform this multiple times. FYI, I have also IKEA E1743 on/off switch and it is facing the same exception after "Configuration 0x00xxxxx" step.

Debug logs:
[https://pastebin.com/ae7daKgk](url)

Koenkk, 2020-04-22:
I don't understand why it happens yet. Perhaps sniffing reveals more information: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Note that the configure is not necessary for this device to work properly. It is only used to retrieve the linkquality.

closed 2020-07-02 by stale[bot]