Koenkk/zigbee2mqtt #3411: Zigbee map not working with Xiaomi

aureliosaraiva, 2020-04-22:
# Bug Report
## What happened

Guys, when building a map of my zigbee network. I can see the devices linked to a router or coordinator.

But Xiaomi devices don't show a link between them, I don't know which device they are connected to.

Yes, they are online, at HomeAssistant I can receive an update signal from them and they are working with my automations.

## What did you expect to happen

I hope you will show me the route he is connected to and which he shares.

## How to reproduce it (minimal and precise)

[image]

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version: Zigbee2mqtt 1.12.2

Koenkk, 2020-04-23:
Xiaomi devices are sleeping most of the time actually so it happens that they show disconnected on the network map. Triggering the device manually might provide a temporary workaround.

sjorge, 2020-04-23:
> Xiaomi devices are sleeping most of the time actually so it happens that they show disconnected on the network map. Triggering the device manually might provide a temporary workaround.

Is that also why they do show up with the source routing firmware? (coordinator or a router responds in there place?)

aureliosaraiva, 2020-04-23:
> Xiaomi devices are sleeping most of the time actually so it happens that they show disconnected on the network map. Triggering the device manually might provide a temporary workaround.

I understand, then is a behavior normal?

Koenkk, 2020-04-24:
> > Xiaomi devices are sleeping most of the time actually so it happens that they show disconnected on the network map. Triggering the device manually might provide a temporary workaround.
> 
> I understand, then is a behavior normal?

yes

closed 2020-07-02 by stale[bot]