Koenkk/zigbee2mqtt #3412: I fail to set my Xiaomi B1 curtain motor to reverse mode.

cchalong, 2020-04-23:
# Bug Report
## What happened
1. as guided, I tried to publish mqtt msg with payload for reserve mode
**{"options": {"reverse_direction":"true"}}**
2. but I encountered error msg in zigbee2mqtt log(add on)
**zigbee2mqtt:error 2020-04-23 11:05:30: No converter available for 'options' ([object Object])**

I use Hassio and zigbee2mqtt add on which are updated with latest version.

## What did you expect to happen
I wanted to use reverse mode as my b1 motor showed "open" when the curtain is closed in reality.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

Koenkk, 2020-04-23:
This was not enabled for the B1 curtain motor yet, just did it please try the latest edge addon in 5 hours from now.

cchalong, 2020-04-24:
it seems that I lost the opportunity to try in 5 hours because I noticed your comment too late.

at the moment, even though I amended the devices.js in my docker, same error msg came out.

nukusinji, 2020-04-24:
I tried latest dev, Option is working. but command is failed.
I found command packet is different between Xiaomi motor.

B1 packet is here https://github.com/freenetwork/ZNCLDJ12LM
Command is different Attribute, data type, ETC.

Please check packet. Thanks

Koenkk, 2020-04-26:
@nukusinji thanks, I've implemented this.

Changes will be available in the latest dev brach in 3 hours from now.
Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/ZNCLDJ12LM.md#configuration-of-device-attributes

cchalong, 2020-04-26:
I still have a problem after amending 2 files, tozigbee.js and devices.js as you guided.

the error msg in zigbee2mqtt is as follow

zigbee2mqtt:error 2020-04-26 21:19:00: Publish 'set' 'options' to 'xiaomi_curtain_b1' failed: 'Error: Write 0x04cf8cdf3c7559c9/1 genBasic({"65320":{"value":"true","type":1}}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Write for 'BUFFER' not available)'

Thanks for your checking again

nukusinji, 2020-04-26:
> @nukusinji thanks, I've implemented this.
> 
> Changes will be available in the latest dev brach in 3 hours from now.
> Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/ZNCLDJ12LM.md#configuration-of-device-attributes

Thanks.
But I still have error.
zigbee2mqtt:error 2020-04-26 22:24:10: Publish 'set' 'options' to 'curtain_inner_living_room' failed: 'Error: Write 0x04cf8cdf3
c73ed61/1 genBasic({"1025":{"value":[7,0,2,0,1,4,0,18],"type":66}}, {"timeout":10000,"manufacturerCode":4447,"disableDefaultRes
ponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')'  
I thinks the packet is older version.
We have to wait when some one provide new packet data because I don't have gateway.

nukusinji, 2020-04-27:
> @nukusinji thanks, I've implemented this.
> Changes will be available in the latest dev brach in 3 hours from now.
> Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/ZNCLDJ12LM.md#configuration-of-device-attributes

I hope, this link help for you. (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1654#issuecomment-523061659)

please support battery level in z2m.

Koenkk, 2020-04-28:
I forgot to update after this, please try again with latest dev branch.

nukusinji, 2020-04-28:
> I forgot to update after this, please try again with latest dev branch.

Thanks. I try latest dev and I saw error.

Follow the debug logs.

zigbee2mqtt:debug 2020-04-29 01:09:02: Received MQTT message on 'Ryan_home/zigbee2mqtt/curtain_inner_living_room/set' with data '{                                                                          
    "options":{                                                                                                                                                                                             
        "reverse_direction": true,                                                                                                                                                                          
        "auto_close": true                                                                                                                                                                                  
    }                                                                                                                                                                                                       
}'                                                                                                                                                                                                          
zigbee2mqtt:debug 2020-04-29 01:09:02: Publishing 'set' 'options' to 'curtain_inner_living_room'                                                                                                            
zigbee2mqtt:error 2020-04-29 01:09:02: Publish 'set' 'options' to 'curtain_inner_living_room' failed: 'Error: Write 0x04cf8cdf3c73ed61/1 genBasic({"65320":{"value":true,"type":1}}, {"timeout":10000,"manuf
acturerCode":null,"disableDefaultResponse":true}) failed (Error: Write for 'BUFFER' not available)'                                                                                                         
zigbee2mqtt:debug 2020-04-29 01:09:02: Error: Write 0x04cf8cdf3c73ed61/1 genBasic({"65320":{"value":true,"type":1}}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error:
 Write for 'BUFFER' not available)                                                                                                                                                                          
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)                                                                                                    
    at Generator.throw (<anonymous>)                                                                                                                                                                        
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)                                                                                                                  
    at runMicrotasks (<anonymous>)                                                                                                                                                                          
    at processTicksAndRejections (internal/process/task_queues.js:97:5)                                                                                                                                     
zigbee2mqtt:info  2020-04-29 01:09:02: MQTT publish: topic 'Ryan_home/zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'options' to 'curtain_inner_living_room' fai
led: 'Error: Write 0x04cf8cdf3c73ed61/1 genBasic({\"65320\":{\"value\":true,\"type\":1}}, {\"timeout\":10000,\"manufacturerCode\":null,\"disableDefaultResponse\":true}) failed (Error: Write for 'BUFFER' n
ot available)'","meta":{"friendly_name":"curtain_inner_living_room"}}'

Koenkk, 2020-04-28:
Can you check if this has been fixed in the latest dev branch?

nukusinji, 2020-04-29:
> Can you check if this has been fixed in the latest dev branch?

It's same. 

But I found where is bug.
variable name has error.

            const options = {
                'reverse_direction': false,
                'auto_close': true,
                ...value,
            };

I try to change to 'opt1s' from 'options'. It saw working. 
But I try remote connection. So, I didn't know Weather It work right cmd, 
However, I think it is right because room brightness is change by curtain move direction.
I can notify in 8 hours.

nukusinji, 2020-04-29:
I found check method for battery level.
https://github.com/uqpqpu/st-driver/blob/master/devicetypes/shinjjang/xiaomi-curtain-b1.src/xiaomi-curtain-b1.groovy

I don't know how to apply async read command. 
read command Attribute value is 0x0021  

Thanks for update.

cchalong, 2020-04-29:
> > Can you check if this has been fixed in the latest dev branch?
> 
> It's same.
> 
> But I found where is bug.
> variable name has error.
> 
> ```
>         const options = {
>             'reverse_direction': false,
>             'auto_close': true,
>             ...value,
>         };
> ```
> 
> and have to delete line 1116
> 1116: return {state: {options}};
> 
> I try to change to 'opt1s' from 'options'. It saw working.
> But I try remote connection. So, I didn't know Weather It work right cmd,
> However, I think it is right because room brightness is change by curtain move direction.
> I can notify in 8 hours.

where should I change the name to opt1s?
there are too many options in the files. 

and I still failed to set reverse_direction.
the error msg is as follows.

zigbee2mqtt:error 2020-04-29 23:41:39: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined
    at /zigbee2mqtt-1.12.0/lib/extension/entityPublish.js:146:56
    at Array.find (<anonymous>)
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.12.0/lib/extension/entityPublish.js:146:42)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.12.0/lib/controller.js:346:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.12.0/lib/controller.js:242:14)
    at MQTT.emit (events.js:223:5)
    at MQTT.onMessage (/zigbee2mqtt-1.12.0/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:223:5)
    at MqttClient._handlePublish (/zigbee2mqtt-1.12.0/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/zigbee2mqtt-1.12.0/node_modules/mqtt/lib/client.js:351:12))

nukusinji, 2020-04-29:
> It's same.
>
> But I found where is bug.
> variable name has error.
>
> ```
>         const options = {
>             'reverse_direction': false,
>             'auto_close': true,
>             ...value,
>         }; [...]

Working is Good! Thanks ~~~~

Koenkk, 2020-04-29:
Fixed in latest dev branch.

cchalong, 2020-04-29:
> Fixed in latest dev branch

Appreciate!, Koenkk and nukusinji.
I finally succeeded in setting to reverse_direction.

by the way, it there any ways to use latest dev branch when using add-on in Hass.io?
I just amended the files inside the docker.

Koenkk, 2020-04-30:
@cchalong use the edge version of the addon.

Assuming this can be closed now.

closed 2020-04-30 by Koenkk

anhtuanng98, 2020-05-17:
> # Bug Report
> ## What happened
> 1. as guided, I tried to publish mqtt msg with payload for reserve mode
>    **{"options": {"reverse_direction":"true"}}**
> 2. but I encountered error msg in zigbee2mqtt [...]

Can you tell me how to set reverse_direction to xiaomi curtain? I dont understand how to config `{
    "options":{
        "reverse_direction": xxx,
        "hand_open": xxx,
        "reset_move": xxx
    }
}`

Thank you so much

mpuff, 2020-05-21:
i have canged the reverse_direction, but ow can i do the recalibration?

anhtuanng98, 2020-05-21:
> i have canged the reverse_direction, but ow can i do the recalibration?
> 
> 

Fully open and close by your hand

mpuff, 2020-05-21:
i cant fully open and close, it has saved the position (edges), is there a way to reset it with the button?

anhtuanng98, 2020-05-21:
> i cant fully open and close, it has saved the position (edges), is there a way to reset it with the button?
> 
> 
> 
> when i do it by hand, it is not doing the recalibration, so the reverse_direction is not taken affect.

Open plastic part under this motor like this video.

https://youtu.be/3lqIXm73qbk

Goodluck to you