Koenkk/zigbee2mqtt #3415: Adding new device - how to do more debugging?

swissbyte, 2020-04-23:
Hi there.

I would like to add my own (DIY) device to zigbee2mqtt. For this i have enabled all possible debugging features. Now when trying to pair i get the following output: 

```
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,1,95,1,95,215,236,177,20,0,75,18,0,128,206] +1m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,1,95,1,95,215,236,177,20,0,75,18,0,128,206] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [1,95,1,95,215,236,177,20,0,75,18,0,128] - 206 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":24321,"nwkaddr":24321,"ieeeaddr":"0x00124b0014b1ecd7","capabilities":128} +1m
  zigbee-herdsman:controller:log Device announce '0x00124b0014b1ecd7' +1m
  zigbee-herdsman:controller:log Device announce is from unknown device '0x00124b0014b1ecd7' +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:controller:log Permit joining +2m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +3ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +3ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
zigbee2mqtt:debug 2020-04-23 18:58:56: Saving state to file C:\code\node.js\zigbee2mqtt\data\state.json
  zigbee-herdsman:controller:log Permit joining +3m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,2,4,194,118,1,1,0,39,0,175,205,50,0,0,8,24,123,10,0,0,41,2,9,194,118,29,246] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,2,4,194,118,1,1,0,39,0,175,205,50,0,0,8,24,123,10,0,0,41,2,9,194,118,29,246] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,2,4,194,118,1,1,0,39,0,175,205,50,0,0,8,24,123,10,0,0,41,2,9,194,118,29] - 246 +2ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1026,"srcaddr":30402,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":39,"securityuse":0,"timestamp":3329455,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,123,10,0,0,41,2,9]}} +3m
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":123,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":2306}]},"address":30402,"endpoint":1,"linkquality":39,"groupID":0}' +3m
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '30402', skipping... +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,5,4,194,118,1,1,0,39,0,187,205,50,0,0,8,24,124,10,0,0,33,248,9,194,118,29,16,254,37,68,129,0,0,3,4,194,118,1,1,0,39,0,198,205,50,0,0,17,24,125,10,0,0,41,195,3,20,0,40,255,16,0,41,158,37,194,118,29,50] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,5,4,194,118,1,1,0,39,0,187,205,50,0,0,8,24,124,10,0,0,33,248,9,194,118,29,16,254,37,68,129,0,0,3,4,194,118,1,1,0,39,0,198,205,50,0,0,17,24,125,10,0,0,41,195,3,20,0,40,255,16,0,41,158,37,194,118,29,50] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,5,4,194,118,1,1,0,39,0,187,205,50,0,0,8,24,124,10,0,0,33,248,9,194,118,29] - 16 +2ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1029,"srcaddr":30402,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":39,"securityuse":0,"timestamp":3329467,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,124,10,0,0,33,248,9]}} +10ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":124,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":33,"attrData":2552}]},"address":30402,"endpoint":1,"linkquality":39,"groupID":0}' +6ms
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '30402', skipping... +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,37,68,129,0,0,3,4,194,118,1,1,0,39,0,198,205,50,0,0,17,24,125,10,0,0,41,195,3,20,0,40,255,16,0,41,158,37,194,118,29,50] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 37 - 2 - 4 - 129 - [0,0,3,4,194,118,1,1,0,39,0,198,205,50,0,0,17,24,125,10,0,0,41,195,3,20,0,40,255,16,0,41,158,37,194,118,29] - 50 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1027,"srcaddr":30402,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":39,"securityuse":0,"timestamp":3329478,"transseqnumber":0,"len":17,"data":{"type":"Buffer","data":[24,125,10,0,0,41,195,3,20,0,40,255,16,0,41,158,37]}} +4ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":125,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":963},{"attrId":20,"dataType":40,"attrData":-1},{"attrId":16,"dataType":41,"attrData":9630}]},"address":30402,"endpoint":1,"linkquality":39,"groupID":0}' +4ms
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '30402', skipping... +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +7ms
zigbee2mqtt:debug 2020-04-23 19:03:55: Saving state to file C:\code\node.js\zigbee2mqtt\data\state.json
  zigbee-herdsman:controller:log Permit joining +47s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +3m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82] +47s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +47s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
```

I dont have any idea why i dont get any manufacuter infos out of my device. 
Maybe someone else can see a problem?
Or has any other idea how to debug this further more?

Thanks. 
Regards

Koenkk, 2020-04-24:
Please share the herdsman debug logging when pairing the device.

swissbyte, 2020-04-27:
Thank you for your answer. 
The problem was in fact my own pairing routine. The device is now able to pair.

closed 2020-04-27 by swissbyte