Koenkk/zigbee2mqtt #3416: Configuring of Osram Smart+ Motion is failing

Kosta-Github, 2020-04-23:
# Bug Report

Trying to pair and configure an Osram Smart+ Motion sensor ()

## What happened

In the logs I can see this error:

```
Successfully interviewed '0x000d6f000f77e5ed', device has successfully been paired
zigbee2mqtt    | 2020-04-23T19:46:23.327811354Z zigbee2mqtt:info  2020-04-23 21:46:23: Device '0x000d6f000f77e5ed' is supported, identified as: OSRAM SMART+ Motion Sensor (AC01353010G)
zigbee2mqtt    | 2020-04-23T19:46:23.331025044Z zigbee2mqtt:info  2020-04-23 21:46:23: Configuring '0x000d6f000f77e5ed'
zigbee2mqtt    | 2020-04-23T19:46:23.339906219Z zigbee2mqtt:info  2020-04-23 21:46:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f000f77e5ed","model":"AC01353010G","vendor":"OSRAM","description":"SMART+ Motion Sensor","supported":true}}'
zigbee2mqtt    | 2020-04-23T19:46:33.381895961Z zigbee2mqtt:error 2020-04-23 21:46:33: Failed to configure '0x000d6f000f77e5ed', attempt 1 (Error: ConfigureReporting 0x000d6f000f77e5ed/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt    | 2020-04-23T19:46:33.382123773Z     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
zigbee2mqtt    | 2020-04-23T19:46:33.382186117Z     at Generator.next (<anonymous>)
zigbee2mqtt    | 2020-04-23T19:46:33.382221534Z     at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
zigbee2mqtt    | 2020-04-23T19:46:44.653493286Z zigbee2mqtt:info  2020-04-23 21:46:44: MQTT publish: topic 'zigbee2mqtt/0x000d6f000f77e5ed', payload '{"occupancy":false,"tamper":true,"battery_low":false,"linkquality":52}'
zigbee2mqtt    | 2020-04-23T19:46:44.662617274Z zigbee2mqtt:info  2020-04-23 21:46:44: Configuring '0x000d6f000f77e5ed'
zigbee2mqtt    | 2020-04-23T19:46:54.742124738Z zigbee2mqtt:error 2020-04-23 21:46:54: Failed to configure '0x000d6f000f77e5ed', attempt 2 (Error: ConfigureReporting 0x000d6f000f77e5ed/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt    | 2020-04-23T19:46:54.742503227Z     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
zigbee2mqtt    | 2020-04-23T19:46:54.742691820Z     at Generator.next (<anonymous>)
zigbee2mqtt    | 2020-04-23T19:46:54.742765570Z     at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

## What did you expect to happen

No errors :-)

## How to reproduce it (minimal and precise)

Started pairing with that device.

In order to actually being able to start pairing I had to switch to `channel: 25` in the config; all other channels were not working at all. No attempt to pair appeared in the logs. Is this expected?

## Debug Info
Zigbee2mqtt version: 1.22.2
Adapter hardware: CC2531
Adapter firmware version: zStack12, 20190608

```
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
```

Koenkk, 2020-04-24:
Try holding the device close to the coordinator, trigger a configure through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and right before doing that wake up by clicking a button on it.

Kosta-Github, 2020-04-27:
I paired successfully with `5` of these motion detectors, but all of them are showing the same errors.

I assume, that either the `batteryPercentageRemaining` property is not supported by them, or that this property might need to be spelled (a bit) differently for them?

Koenkk, 2020-04-28:
Can you try changing it to:

```js
{
    zigbeeModel: ['Motion Sensor-A'],
    model: 'AC01353010G',
    vendor: 'OSRAM',
    description: 'SMART+ Motion Sensor',
    supports: 'occupancy, tamper and temperature',
    fromZigbee: [
        fz.temperature,
        fz.ias_occupancy_alarm_2,
    ],
    toZigbee: [],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await configureReporting.temperature(endpoint);
        await configureReporting.batteryVoltage(endpoint);
    },
},
```

Schmidda83, 2020-04-29:
Hi Koenkk,

first thanks for peace of Software, great work.
I'm totally new and I'm currently following a bunch of turtorials for my Smart Home.

I'm facing the same issue as Kosta above. 

You're saying that I shall change something, but my question is

   Which file or configuration?

For this doesn't look like the configration.yaml, or am I wrong?

Koenkk, 2020-04-29:
In `devices.js`, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to get to that file.

Schmidda83, 2020-04-29:
zigbee2mqtt:info  2020-04-29 21:13:56: Successfully configured 'Motion_Sensor_Eingang_1'

thx Koenkk

Koenkk, 2020-05-02:
Merged, assuming this can be closed.

closed 2020-05-02 by Koenkk