Koenkk/zigbee2mqtt #3417: Problem Innr RF 263

Openhabuser, 2020-04-23:
# Bug Report
## What happened
I have add an Innr RF 263 light bulb to Zigbee2Mqtt. I am able to switch the state between on and off. But I can not change the brightness. The brightness is missing in the payload:
Apr 23 21:58:24 openhab npm[8541]: zigbee2mqtt:info  2020-04-23 21:58:24: MQTT publish: topic 'zigbee2mqtt/Innr_Bulb_1', payload '{"state":"ON"}'

## What did you expect to happen
I expect that I can change the brightness like it is also described in the supported device section.

## How to reproduce it (minimal and precise)
-

## Debug Info
Zigbee2mqtt version: 1.12.2
Adapter hardware: CC2531
Adapter firmware version: revision 20190608

sjorge, 2020-04-23:
Have you tried setting a brightness? Did it show an error after setting it?

Openhabuser, 2020-04-23:
Yes, I also tried to change the brightness with OpenHab.
[image]

sjorge, 2020-04-23:
I guess it never makes it to z2m, can you try setting it once directly via mqtt?

mosquitto_pub -h localhost -p 1883 -t zigbee2mqtt/Innr_Bulb_1/set -m '{"brightness_percent": 100}'

Adapt host/port/... where needed.

Openhabuser, 2020-04-24:
Thanks for your hint.

It is very confusing for me.
I tried to log the MQTT Broker with mqtt.fx.
No the payload of the bulb changed:
Apr 24 07:43:09 openhab npm[8541]: zigbee2mqtt:info  2020-04-24 07:43:09: MQTT publish: topic 'zigbee2mqtt/Innr_Bulb_1', payload '{"state":"ON","brightness":100}'

How is this possible? Also the "transition" is missing.

Any idea?

sjorge, 2020-04-24:
AFAIK transition is never published, I’m guessing if you have reporting off, there will be no property for brightness/color/... until you set something first. Or if you do, the bulb may not support reporting. (e.g. hue)

~ sjorge

closed 2020-07-02 by stale[bot]