Koenkk/zigbee2mqtt #3419: Error: Timeout - 43342 - 11 - 2 - 6 - 7 after 6000ms with 9290012573A - Philips Hue white and color ambiance E26/E27/E14

gelsas, 2020-04-23:
# Bug Report
## What happened

Paired the device successfully:

```
Apr 24 01:27:23 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:23: Device '0x001788010278badc' joined
Apr 24 01:27:23 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:23: Starting interview of '0x001788010278badc'
Apr 24 01:27:23 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x001788010278badc"}}'
Apr 24 01:27:23 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x001788010278badc"}}'
Apr 24 01:27:23 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x001788010278badc"}}'
Apr 24 01:27:32 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:32: Successfully interviewed '0x001788010278badc', device has successfully been paired
Apr 24 01:27:32 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:32: Device '0x001788010278badc' is supported, identified as: Philips Hue white and color ambiance E26/E27/E14 (9290012573A)
Apr 24 01:27:32 openhab npm[3545]: zigbee2mqtt:info  2020-04-24 01:27:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x001788010278badc","model":"9290012573A","vendor":"Philips","description":"Hue white and color ambiance E26/E27/E14","supported":true}}'

```
Then the next thing in the log is:

```
Apr 24 01:28:02 openhab npm[3839]: zigbee2mqtt:error 2020-04-24 01:28:02: Failed to setup reporting for '0x001788010278badc' - Error: ConfigureReporting 0x001788010278badc/11 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 47266 - 11 - 2 - 6 - 7 after 6000ms)
Apr 24 01:28:02 openhab npm[3839]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:326:23)
Apr 24 01:28:02 openhab npm[3839]:     at Generator.throw (<anonymous>)
Apr 24 01:28:02 openhab npm[3839]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)

```

## What did you expect to happen
Reporting to be setup correctly.

## How to reproduce it (minimal and precise)
Pair a 9290012573A - Philips Hue white and color ambiance E26/E27/E14 blub

## Debug Info
zigbee2mqtt version: `zigbee2mqtt version 1.12.2 (commit #911cd58)`
CC1352P-2
`Apr 24 01:27:47 openhab npm[3839]: zigbee2mqtt:info  2020-04-24 01:27:47: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.2","commit":"911cd58","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"info","permit_join":true}'`

sjorge, 2020-04-24:
Just hit this again too while testing some stuff, sadly this is expected as an up-to-date (firmware wise) Hue bulb no longer support reporting for genOnOff.

z2m should probably only try to set it up on older firmware.

gelsas, 2020-04-24:
@sjorge 
True z2m should check firmware first and only then set it up if old firmware is detected.

Is the bulb reporting the linkquality for you atleast? 
Or does it do no reporting anything at all like mine ?

(Unreleated question: Did you ever try to bind a dimmer to such a bulb ? - I just tried that and get a whole set of errors thrown out by z2m. https://github.com/Koenkk/zigbee2mqtt/issues/2100#issuecomment-619029676)

sjorge, 2020-04-24:
Yes, it was reporting lqi for me: `{"state":"ON","brightness":127, "linkquality":5}`

I don't have the dimmer paired currently as I switched the little ikea on/off remotes. But I did have it working yes, but I think I was using groups at that point.

So dimmer bind to group, bulb add to group

gelsas, 2020-04-24:
But now after you hit the same error as I it did not report it anymore ? 
(Just want to be sure I understood it correctly, because for me it is currently not reporting the linkquality etc.)

(Unrelated question again: 
I want to try the way you suggested to bind it to a group but want to make sure I understand the workflow correctly.
Bind Dimmer Switch to group --> Add bulb to group --> Unbind dimmer from coordinator --> Done.

sjorge, 2020-04-24:
You unbind the dimmer form the coordinator first, easy way to remember is...

you can't put the coffee in the cup (bind) if there is already tea in it, poor out the tea (unbind) first.

gelsas, 2020-04-24:
Thank you for the help.

The documentation for the dimmer switch shows this:
[...] and unbind it from the coordinator. Use the dimmer as source and a literal coordinator as target for that.

Thats what I just tried but it also results in an error sadly:

```
[16:04:12] openhabian@openhab:~$ mosquitto_pub -t zigbee2mqtt/bridge/unbind/HueDimmerSwitchWandlampe -m coordinator
[16:33:34] openhabian@openhab:~$ sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at Thu 2019-02-14 11:11:58 CET. --
Apr 24 16:33:44 openhab npm[29634]: zigbee2mqtt:error 2020-04-24 16:33:44: Failed to unbind cluster 'genScenes' from 'HueDimmerSwitchWandlampe' to 'Coordinator' (Error: Unbind 0x00178801020e4808/1 genScenes from '0x00124b001ca1aefd/1' failed (Error: AREQ - ZDO - unbindRsp after 10000ms))
Apr 24 16:33:44 openhab npm[29634]: zigbee2mqtt:info  2020-04-24 16:33:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"Coordinator","cluster":"genScenes"}}'
Apr 24 16:33:54 openhab npm[29634]: zigbee2mqtt:error 2020-04-24 16:33:54: Failed to unbind cluster 'genOnOff' from 'HueDimmerSwitchWandlampe' to 'Coordinator' (Error: Unbind 0x00178801020e4808/1 genOnOff from '0x00124b001ca1aefd/1' failed (Error: AREQ - ZDO - unbindRsp after 10000ms))
Apr 24 16:33:54 openhab npm[29634]: zigbee2mqtt:info  2020-04-24 16:33:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"Coordinator","cluster":"genOnOff"}}'
Apr 24 16:34:04 openhab npm[29634]: zigbee2mqtt:error 2020-04-24 16:34:04: Failed to unbind cluster 'genLevelCtrl' from 'HueDimmerSwitchWandlampe' to 'Coordinator' (Error: Unbind 0x00178801020e4808/1 genLevelCtrl from '0x00124b001ca1aefd/1' failed (Error: AREQ - ZDO - unbindRsp after 10000ms))
Apr 24 16:34:04 openhab npm[29634]: zigbee2mqtt:info  2020-04-24 16:34:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind_failed","message":{"from":"HueDimmerSwitchWandlampe","to":"Coordinator","cluster":"genLevelCtrl"}}'

```

Is this to much offtopic? Should I open a seperate issue for this?

closed 2020-07-02 by stale[bot]