Koenkk/zigbee2mqtt #3421: Coordinator ieeeAddr sometimes shows up in data/configuration.yaml

sjorge, 2020-04-24:
# Bug Report
## What happened

Sometimes I have an entry show up for the coordinator in data/configuration.yaml under the devices section.

```
  '0x00124b001938a7e5':
    friendly_name: '0x00124b001938a7e5'
```

```
[root@amethyst /opt/zigbee2mqtt]# grep '"ieeeAddr":"0x00124b001938a7e5"' data/database.db
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001938a7e5","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[]},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1587499244203}
```

## What did you expect to happen
Not to have the coordinator show up there.

## How to reproduce it (minimal and precise)
Not sure, I've seen it twice... but I am unable to reproduce it on demand.
I think it might have something to do with a device removal failing... but that is just the best guess given at least one time i think it show up after removing a device that failed and had to be force removed.

## Debug Info
Zigbee2mqtt version: 1.12.2-dev (commit #63ecc15)
Adapter hardware: CC2531
Adapter firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20200410}}`

Koenkk, 2020-04-24:
Should be fixed in latest dev, thanks!

sjorge, 2020-04-24:
👍 that was quick!

closed 2020-04-24 by sjorge