Koenkk/zigbee2mqtt #3424: IKEA E1743 - Unbind from default_bind_group

tsprivate, 2020-04-25:
Hey I got a few Ikea E1743 switches and they easily get detected.

Unfortunately I'm unable to UNBIND the devices from "default_bind_group" via MQTT command (Topic - zigbee2mqtt/bridge/bind/{friendly_name}). I always get response UNBIND failed.

- Is there any way to avoid the BIND to the "default_bind_group" ?
- how to properly UNBIND

Thanks for any help 

## Debug Info
Zigbee2mqtt version:  1.12.2
Adapter hardware:  CC26X2R1
Adapter firmware version: 3.x.0
Device count: 36

Anjerlaan, 2020-04-25:
Did you wake the device up just before sending the unbind command?

tsprivate, 2020-04-25:
Yes, as well tried kept pressing while sending :(

Anjerlaan, 2020-04-26:
What “friendly” group name did you use?

tsprivate, 2020-04-26:
I use the "default_bind_group" as instructed [here](https://www.zigbee2mqtt.io/devices/E1743.html)

didiht, 2020-04-27:
I think they have got different group Id's during pairing.
You have to find out the **groupId** in the log file (debug level) and create (temporary) a new group with this **groupId**.
Now you should be able to unbind (from the new group) and bind to another group.
Hope this helps.

Anjerlaan, 2020-04-27:
That was next suggestion...I also read somewhere that 'coordinator' as payload could also work

tsprivate, 2020-04-27:
I have tried both. 

- With 'coordinator' I get the unbind_failed response.

- I could only find the groupID '0' in the log, but when unbind from '0' there is no response at all from z2m.

didiht, 2020-04-28:
In my opinion groupId '0' indicates that the switch is not bound.
You should be able to bind to another group?
If you still have trouble you can try to update the firmware of the switch.

tsprivate, 2020-04-28:
The switch has already updated. 

The switch must be bound to a group as my switchable Power-Sockets (xiaomi) switch on/off according the IKEA switch button pressed

ralphhendriks, 2020-05-12:
I just acquired the TRÅDFRI set consisting of one E27 860 lm bulb (LED1836G9) and the E1743 remote. I first paired the bulb with zigbee2mqtt, then paired the remote with zigbee2mqtt, and then tried to set up binding.
As described in the docs, the unbind from the `default_bind_group` seems to work okay.
However, I did not succeed to bind the remote (source) directly to the bulb (remote). I have tried a number of variations of the command, did a lot of re-pairing, etc., all to no avail.
Then I created a new group, added the bulb to the group, and paired the remote to the group. This worked flawlessly. Maybe the remote does not support binding to a single device directly? Or is this a shortcoming/bug in zigbee2mqtt.
Anyway, maybe this approach might be of help.

Nauttor, 2020-07-15:
Hi, 

I'm on the same boat here, I have experienced the same unreliable behave from the unbinding/binding procedure. I have 4 devices that does not work as expected when trying to unbind/bind them. All of them are on the last Firmware version. The cc2531 is loaded with the last 1.2 and the z2m version is the last from GitHub. 

E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
E1525/E1745 - IKEA TRADFRI motion sensor (EndDevice)
ZYCT-202 - Trust Remote control (EndDevice)

The first two initially appears to be in the group 901. No matter what I tried they didn't leave the group. Only after I create the 901 group by hand, bind them both and then unbind, they start working on the new group. However the process wasn't that straight, was more a hit and miss, I try different orders and sometimes it works as expected and sometimes not. Now they are on group 9, but I'm sure If I remove them I will be on the same situation again.  

About the other two devices, the motion sensor is on group 7787(randomly changed every time I remove and add it) and the Trust on 150, no matter the procedure I follow, they stay in their group and I can't unbind them. I tried the same procedure to create the 7787 and 150 to bind and unbind, don't work. BTW, yes I wake them every time. Funny thing, it always says successful unbinding / binding, but it continues in the same group. 

Unbinding 

Jul 15 14:29:43 atom npm[4866]: #033[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:29:43: Received MQTT message on 'zigbee2mqtt/bridge/unbind/0x000d6ffffe2672bd' with data '7787'
Jul 15 14:29:43 atom npm[4866]: #033[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:29:43: unbinding cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'tele'
Jul 15 14:29:46 atom npm[4866]: #033[32mzigbee2mqtt:info #033[39m 2020-07-15 14:29:46: Successfully unbound cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'tele'
Jul 15 14:29:46 atom npm[4866]: #033[32mzigbee2mqtt:info #033[39m 2020-07-15 14:29:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind","message":{"from":"0x000d6ffffe2672bd","to":"tele","cluster":"genOnOff"}}'

Binding

[34mzigbee2mqtt:debug#033[39m` 2020-07-15 14:25:54: Received MQTT message on 'zigbee2mqtt/bridge/bind/0x000d6ffffe2672bd' with data 'g_2'
[39m 2020-07-15 14:25:54: binding cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'g_2'
[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:25:56: Received Zigbee message from '0x000d6ffffe2672bd', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"ontime":600,"offwaittime":0}' from endpoint 1 with groupID 7787
[39m 2020-07-15 14:25:56: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2672bd', payload '{"occupancy":true,"linkquality":65,"update_available":false}'
[32mzigbee2mqtt:info #033[39m 2020-07-15 14:25:57: Successfully bound cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'g_2'
[32mzigbee2mqtt:info #033[39m 2020-07-15 14:25:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"0x000d6ffffe2672bd","to":"g_2","cluster":"genOnOff"}}'
[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:26:00: Received Zigbee message from '0x000d6ffffe2672bd', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"ontime":600,"offwaittime":0}' from endpoint 1 with groupID 7787
[32mzigbee2mqtt:info #033[39m 2020-07-15 14:26:00: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2672bd', payload '{"occupancy":true,"linkquality":63,"update_available":false}'